warpem/warp #181: WarpTools generated tomograms may not be suitable for high resolution work

alisterburt, 2024-07-10:
There are a couple of pieces of evidence pointing at something being off with tomogram generation in `WarpTools`

1. inability to reproduce template matching peaks at ~14 SD for EMPIAR-10499
2. inability to refine particles extracted from WarpTools generated tomograms in Dynamo

Particles generated via `ts_export_particles` still refine to high res. This issue breaks the Warp-Dynamo-M workflow some people like.

I saw the 14SD mentioned in (1) with an earlier version of WarpTools. (2) was reported by Viraj Upadhye.
There have been a few reports of the tomograms 'looking fuzzy' and people being unhappy with template matching.

@dtegunov mentioned he had changed the weighting in Fourier space somewhat recently to fix a bug - perhaps this is the root cause?

alisterburt, 2024-07-11:
dimitry claims everything is as it should be - we haven't discussed in detail as he's away at a meeting

dtegunov, 2024-07-11:
The bug we discussed was fixed several years ago. The most likely source of <14 SD peaks in 10499 is different initial alignment.

alisterburt, 2024-07-11:
So that it's written down:

- when doing subvolume reconstruction values from 2D data are inserted into a 3D transform
- values are 'spread' across the 8 nearest voxels with linear interpolation weights
- total weights are kept track of through the reconstruction process
- values where weights are positive are divided by the total weight to 'equalise' the power of all components

what do we do for voxels where weights are <1?

We used to divide by the weight, now we don't

closed 2024-07-14 by dtegunov