warpem/warp #183: ts_etomo_patches using wrong rotation axis and producing invalid preali.mrc file

mpm896, 2024-07-11:
Hello,

I'm running `ts_etomo_patches` with the following command:
```
WarpTools ts_etomo_patches \
--settings warp_tiltseries.settings \
--angpix 4.76 \
--patch_size 500 \ 
--initial_axis 3.00
```
Despite specifying the tilt axis of 3.00, it appears that etomo is being fed a value of -86.68, which is the value from the mdoc. However since this data was collected with Tomography 5, the reported tilt axis is different than serialEM (sigh) and so for the purpose of IMOD, the input tilt axis should be 3 (90 + tilt axis reported from Tomo 5 = 90 - -86.68 ~~ 3).

In addition to this, it appears that `prenewst` is producing an invalid mrc file? I get a log from `tiltseries/logs` with the following output:
```
2024-07-11 14:57:43.701 Received "TomoStack", with 2 arguments, for GPU #0, 21081 MB free:
2024-07-11 14:59:34.010 Created tilt stack for /cloud-data/its-cmo-darwin-magellan-workspaces-folders/WS_Cryoem/CX_LMR/User_directories/Matt/Cryo-ET/PreProcessing_Setup/test_warptools/MM004_g3/100kx/tiltseries/Position_1_2.tomostar
2024-07-11 14:59:34.139 Execution took 110.254 seconds
2024-07-11 14:59:34.282 
2024-07-11 14:59:34.430 Received "TomoEtomoPatchTrack", with 2 arguments, for GPU #0, 21007 MB free:
2024-07-11 14:59:34.671 Performing patch tracking in /cloud-data/its-cmo-darwin-magellan-workspaces-folders/WS_Cryoem/CX_LMR/User_directories/Matt/Cryo-ET/PreProcessing_Setup/test_warptools/MM004_g3/100kx/tiltseries/tiltstack/Position_1_2 with arguments: -DirectiveFile /tmp/tmp5NrKS6.adoc -CurrentLocation /cloud-data/its-cmo-darwin-magellan-workspaces-folders/WS_Cryoem/CX_LMR/User_directories/Matt/Cryo-ET/PreProcessing_Setup/test_warptools/MM004_g3/100kx/tiltseries/tiltstack/Position_1_2 -RootName Position_1_2 -EndingStep 5
2024-07-11 14:59:35.257 To quit all processing, place a Q in the file: /cloud-data/its-cmo-darwin-magellan-workspaces-folders/WS_Cryoem/CX_LMR/User_directories/Matt/Cryo-ET/PreProcessing_Setup/test_warptools/MM004_g3/100kx/tiltseries/tiltstack/Position_1_2/batchruntomo.31592.input
2024-07-11 14:59:35.345 Beginning to process directive file # 1 : /tmp/tmp5NrKS6.adoc   [brt7]
2024-07-11 14:59:35.479 Starting data set Position_1_2   at 14:59:35   [brt1]
2024-07-11 14:59:35.782 
2024-07-11 14:59:57.393 
2024-07-11 14:59:57.566 Batchruntomo started on data set Position_1_2, Thu Jul 11 14:59:35 2024  [brt11]
2024-07-11 14:59:57.708 starting Etomo with log in /root/.etomologs/etomo_err_Jul-11-145937.log
2024-07-11 14:59:57.822 Reached step 0
2024-07-11 14:59:57.937 
2024-07-11 14:59:58.598 Finding coarse alignment by cross-correlation with Tiltxcorr (running xcorr.com)   [brt2]
2024-07-11 15:05:24.271 Successfully finished xcorr.com   in 05:25.7   [brt3]
2024-07-11 15:05:24.451 
2024-07-11 15:05:24.599 Making coarse aligned stack (running prenewst.com)   [brt2]
2024-07-11 15:08:50.319 Successfully finished prenewst.com   in 03:26.0   [brt3]
2024-07-11 15:08:50.505 
2024-07-11 15:08:54.204 Tracking patches to make alignment model (running xcorr_pt.com)   [brt2]
2024-07-11 15:09:24.111 processchunks ERROR: xcorr_pt.com has given processing error 1 times - giving up
2024-07-11 15:09:24.216 ERROR: iiOpen - Position_1_2_preali.mrc has unknown format.
2024-07-11 15:09:24.329 ERROR: iiuOpen - Could not open 'Position_1_2_preali.mrc'
2024-07-11 15:09:24.576  
2024-07-11 15:09:24.738 ABORT SET: An error occurred running xcorr_pt.com
2024-07-11 15:09:24.888 
2024-07-11 15:09:25.066 Batchruntomo finished with data set, Thu Jul 11 15:09:24 2024  [brt12]
2024-07-11 15:09:25.284 
2024-07-11 15:09:25.480 Batch run finished; failures occurred for 1 datasets
2024-07-11 15:09:25.683 Calculating projection parameters from patch tracking results in /cloud-data/its-cmo-darwin-magellan-workspaces-folders/WS_Cryoem/CX_LMR/User_directories/Matt/Cryo-ET/PreProcessing_Setup/test_warptools/MM004_g3/100kx/tiltseries/tiltstack/Position_1_2
2024-07-11 15:09:27.745 Python PID: 3574
2024-07-11 15:09:29.079 Running align.com ... Error executing align.com
2024-07-11 15:09:29.199 ERROR: TILTALIGN - Reading model file: FILE DOES NOT EXIST
2024-07-11 15:09:29.306 ERROR: tiltalign -StandardInput: exited with status 1
2024-07-11 15:09:29.428 Finished for /cloud-data/its-cmo-darwin-magellan-workspaces-folders/WS_Cryoem/CX_LMR/User_directories/Matt/Cryo-ET/PreProcessing_Setup/test_warptools/MM004_g3/100kx/tiltseries/Position_1_2.tomostar
2024-07-11 15:09:29.637 Execution took 595.012 seconds
2024-07-11 15:09:29.781 
```

If I open the etomo gui and generate coarse aligned stack, it works fine and I can open up the resulting coarse aligned stack. However if I try opening the preali.mrc file produced through running `WarpTools ts_etomo_patches ...`, it tells me it's an invalid file format. It's kind of odd, because when I manually generate the coarse aligned stack for this same tilt series, the `prenewst.com` and `prenewst.log` are nearly identical. 

Let me know if you've run into something similar before or have any advice, thanks!

mpm896, 2024-07-11:
Also, is there a way to specify the CPU in WarpTools for parallel processing? ~10 minutes for it to reach this error on a bin 4 tomogram (not even finishing patch tracking) is pretty slow

alisterburt, 2024-07-11:
@mpm896 thanks for the report - I'll look into the tilt axis angle issue and make sure it's being passed down to etomo properly.

It's typically recommended to use 10-15 A/px for patch tracking, you're tracking a lot of patches at 5Apx making them all large, hence the slowdown. When run in the expected way everything runs quickly enough - it's possible IMOD has some environment variables you can set to force things to run the way you want within the wrapper.
You are also not forced to use our wrappers and can run tilt series alignment however you see fit, see [here](https://warpem.github.io/warp/reference/warptools/custom_tilt_series_alignments/) for details

alisterburt, 2024-07-11:
no idea what's going on with your preali file - IMOD creates those internally, we simply call batchruntomo

mpm896, 2024-07-11:
Hey Alister, thanks for the feedback. I agree about the patches - I'll increase the binning or patch size, however it's about 10 minutes to arrive at the patch tracking and then aborting before patch tracking because the preali.mrc is an unknown format. 
Regarding this problem, I'll do a bit more testing to see why `batchruntomo` is outputting a bad preali.mrc and reach out to David Mastronarde with this issue. Thanks!

mpm896, 2024-07-11:
Also meant to ask - to help with this, can you point me to where I can find that .adoc that Warp uses for batchruntomo? The log points to it in `/tmp` but it seems to be getting deleted after the process exits

alisterburt, 2024-07-11:
https://github.com/warpem/warp/blob/afe0bdce289eebe2c67339a180bae18e9895a628/WarpWorker/WarpWorker.cs#L393-L422

🙂

mpm896, 2024-07-11:
Thanks! I had also just found this and also somehow missed the `batchDirective.adoc` in each processing directory

mpm896, 2024-07-11:
Just to update, the issue with speed and my preali file were because I've been doing all my data processing on data stored on an AWS S3 (object storage), which does not behave like a normal file system. Some things work, others do not apparently, and everything is slower. Transferring to a network file system solved these problems

alisterburt, 2024-07-11:
Good to know! thanks again for reporting back

alisterburt, 2024-07-11:
I'm unable to reproduce your error on `2.0.0dev18` 

running 
```sh
WarpTools ts_etomo_patches \
--settings warp_tiltseries.settings \
--angpix 10 \
--patch_size 500 \
--initial_axis 35 \
--output_processing ts_initial_axis_test35
```

yields a `batchdirective.adoc` with the line 

```txt
setupset.copyarg.rotation = 35
```

which matches what the imod documentation says it expects
https://bio3d.colorado.edu/imod/doc/directives.html

and is as bad as expected when the wrong values are used

Going to close based on this test - please reopen if needed

closed 2024-07-11 by alisterburt