warpem/warp #193: Template matching fully covered voxels is too strict

alisterburt, 2024-08-10:
Current criteria are that voxels must be in all tilts to be included - this seems too strict, maybe 80% is better

shahpnmlab, 2024-08-10:
In addition to default thresholds, the option to turn it off would also be good to have (like in the GUI versions)

asarnow, 2024-08-28:
Agree with @shahpnmlab, if we use the tomograms for traditional STA or other 3D analysis there is no need to restrict the matching volume.

shahpnmlab, 2024-08-28:
> Agree with @shahpnmlab, if we use the tomograms for traditional STA or other 3D analysis there is no need to restrict the matching volume.

@asarnow 
My fork for TS matching has this option 
`https://github.com/shahpnmlab/warp/blob/b98b31ef0823c569fa7dd85cdb45136c4dc15bd5/WarpTools/Commands/Tiltseries/TemplateMatchTiltseries.cs#L67`
However, it probably is a good idea to wait until @alisterburt or @dtegunov integrate it into the main branch.

closed 2024-09-10 by alisterburt

dtegunov, 2024-09-10:
--max_missing_tilts in ts_template_match defaults to 2 now.