warpem/warp #194: Question about importing external alignments into warp

shahpnmlab, 2024-08-10:
I have dusted of my imod wrapper and to perform alignments externally and then import those alignments into warp. So far the way the script is constructed is that I am deleting lines from the tomostar file for views/tilts that are badly tracked. however, i still get this error when I tried to import the alignments

```WarpTools ts_import_alignments --settings tiltseries.settings --alignments tiltseries/tiltstack/01122021_BrnoKrios_arctis_lam1_pos5/ --alignment_angpix 10
Running command ts_import_alignments with:
alignments = tiltseries/tiltstack/01122021_BrnoKrios_arctis_lam1_pos5/
alignment_angpix = 10
min_fov = 0
settings = tiltseries.settings
input_data = {  }
input_data_recursive = False
input_processing = null
output_processing = null
input_norawdata = False

No alternative input specified, will use input parameters from tiltseries.settings
File search will be relative to /home/pranav/Desktop/pranav/data/training/11756/data/tomoman_minimal_project/01122021_BrnoKrios_arctis_lam1_pos5/tomostar
1 files found
Parsing previous results for each item, if available...
1/1, previous metadata found for 1
Failed to import alignments for 01122021_BrnoKrios_arctis_lam1_pos5.tomostar: 39 active tilts in series, but 34 lines in /home/pranav/Desktop/pranav/data/training/11756/data/tomoman_minimal_project/01122021_BrnoKrios_arctis_lam1_pos5/tiltseries/tiltstack/01122021_BrnoKrios_arctis_lam1_pos5/01122021_BrnoKrios_arctis_lam1_pos5.xf```

```grep frameseries tomostar/01122021_BrnoKrios_arctis_lam1_pos5.tomostar | wc -l
34```

The only fix I could I come up with was to remove those entries from the xml file as well the <ts_basename>.xml file (setting the UseTilt values to False in the XML for the offending views (but keeping those entries in the tomostar file) does not fix this issue). What would be the way forward? Should I touch both the xml and the tomostar files?

Alternatively, would you consider promoting the tomostar entries during the ts_import_alignemnts step over the XML entries?

Huang-XN, 2024-08-11:
This tutorial might help you.
https://warpem.github.io/warp/reference/warptools/custom_tilt_series_alignments/

alisterburt, 2024-08-23:
Hey! Yeah this isn't ideal - if you run `ts_stack` do you have xml files already or not? If you don't already have XML files we should come up with something... not immediately sure what the best path forwards is though

dtegunov, 2024-08-27:
Is this using etomo's mechanism for tracking removed tilts in a file? A long time ago, I taught desktop Warp to respect the associated file. I'll check if that functionality carried over or needs to be ported.

shahpnmlab, 2024-08-28:
No I am removing them using mrcfile.. (trying to keep the etomo intermediate files to a min)... however if you decide to port that functionality back into WarpTools, I can update my wrappers to be compatible with `excludeviews`...

shahpnmlab, 2024-08-28:
> Hey! Yeah this isn't ideal - if you run `ts_stack` do you have xml files already or not? If you don't already have XML files we should come up with something... not immediately sure what the best path forwards is though

I think XML files are geenrated at ts_stack creation (thats how I could keep track of tilts in the tomostar and the xml files). I managed to make things limp along by modifying both the XML and tomostar files. however, having some clarity on which files (tomostar vs. xml) are important to the workflow would be useful... 

This prompts a more philosophical question - what function does the tomostar file fulfill that the XML doesnt? With the possible exception of frame intensity, almost all the information is logged in (rich) XML files...

alisterburt, 2024-09-03:
@shahpnmlab copying here after our discussion - the `tomostar` files serve as the 'raw data' Warp desktop looks at to determine what might need processing, the corresponding xml is then used which contains richer metadata. 

Dimitry recently added the ability to process from xmls directly 56c467101c9597bb217d867757f5374c2e99e9b0 so things are moving in the direction you want :-)

Also: the functionality for seeing if IMOD removed any views is still present - I don't know if this covers all mechanisms/stages at which IMOD removes images though

https://github.com/warpem/warp/blob/25cd431b82622247def8ec2557feb1aeb2ccddaa/WarpLib/TiltSeries.cs#L1240-L1258

ThatIsMyUsername, 2025-02-17:
Hey, I had the same problem. I'd like to know how the problem was solved. thank you.
`Failed to import alignments for Position_0018.tomostar: 38 active tilts in series, but 39 lines in /data/zcy/mainwarp/xmu-all/AlignTiltSeries/tiltstack/Position_0018/Position_0018.xf`
Path (/data/zcy/mainwarp/xmu-all/AlignTiltSeries/tiltstack/) is the parameters of the **--alignments**.
I looked at the.tomostar file and the.xml file and found that one frame was missing. But I don't know why there is a missing frame, the previous process did not indicate a problem.

mpm896, 2025-04-30:
Hi, I'd like to bump this because I'm getting the same error as @ThatIsMyUsername . I have 41 tilts in the mdoc and 41 xml files in the `frameseries` directory, but only 40 tilts are being written to the `tomostar` file for some tilt series -- the output of `ts_import` shows:

```
Found 1 MDOC files, searching for *.mdoc in /common/workdir/MM039_3/warp/mdoc/
Looking for frame series...
41/41                                                                                                        
Parsing MDOCs and creating .tomostar files...
Successfully parsed Position_2_3_ali.mrc.mdoc, 40 tilts
Successfully parsed 1 MDOC files, 0 failed
```

In my case, I found that the 0-tilt is missing from the tomostar. Let me know if you've found any solutions for this!

ThatIsMyUsername, 2025-05-01:
> Hi, I'd like to bump this because I'm getting the same error as [@ThatIsMyUsername](https://github.com/ThatIsMyUsername) . I have 41 tilts in the mdoc and 41 xml files in the `frameseries` [...]

Sorry, I can't understand your question. If only 40 tilts can be written to tomostar, it is recommended to modify the parameters --c_range_min and --c_range_max.

alisterburt, 2025-05-01:
@mpm896 @ThatIsMyUsername is it possible that some images were deactivated because an earlier program you ran failed on them? 

See WarpTools change_selection for details