warpem/warp #195: Transformation of imod picked coordinates to warp reconstruction

thorstenwagner, 2024-08-12:
Hi,

we have a couple tomograms picked on imod reconstructions. Now we would like to use those coordinates on the warp reconstructions but they don't match. Is there a way to translate the imod coords to warp coordinates?

Best,
Thorsten

thorstenwagner, 2024-08-12:
Here is the tomopitch.log. We think applying half of the ExtraThickness, - ` X axis tilt ` and ` Angle offset ` as y we are visually close.  Would be nice to know the actual transformation from the devs :-) 
```

*** Entries to program tomopitch ***
  ModelFile = tomopitch.mod
  ExtraThickness = 25
  SpacingInY = -1344
  ScaleFactor = 12
  AngleOffsetOld = 12.94
  ZShiftOld = 0.0
  XAxisTiltOld = 0.0
*** End of entries ***

 MODEL FILES OR TIMES MUST OCCUR IN ORDER OF DECREASING SAMPLE COORDINATE

 Analysis of positions from line pair at Y =    57.:
 unrotated lines imply added Z shift of   291.4; thickness of  889.0, set to  896
 slope = -0.0243: to make level, add  -1.4 to total angle offset
  rotated  lines imply added Z shift of   291.4; thickness of  799.4, set to  800

 Analysis of positions from line pair at Y =   171.:
 unrotated lines imply added Z shift of   -55.9; thickness of  889.0, set to  896
 slope = -0.0243: to make level, add  -1.4 to total angle offset
  rotated  lines imply added Z shift of   -55.9; thickness of  799.4, set to  800

 Analysis of positions from line pair at Y =   285.:
 unrotated lines imply added Z shift of  -403.2; thickness of  889.0, set to  896
 slope = -0.0243: to make level, add  -1.4 to total angle offset
  rotated  lines imply added Z shift of  -403.1; thickness of  799.4, set to  800

 Analysis of positions from all line pairs:
 unrotated lines imply added Z shift of   -55.9; thickness of 1583.7, set to 1584
 slope = -0.0243: to make level, add  -1.4 to total angle offset
  rotated  lines imply added Z shift of   -55.9; thickness of 1493.8, set to 1496

 Pitch between samples can be corrected with an added X-axis tilt of  14.28
 In this case, to make level, add  -1.4 to total angle offset
 x-tilted  lines imply added Z shift of   -52.7; thickness of  799.4, set to  800
 
 X axis tilt -  Original:    0.00   Added:   14.28   Total:   14.28
 Angle offset - Original:   12.94   Added:   -1.39   Total:   11.55
 Z shift -      Original:     0.0   Added:   -52.7   Total:   -52.7
SUCCESSFULLY COMPLETED
CHUNK DONE

```

thorstenwagner, 2024-08-12:
Left warp, right imod with fiducials and applied parameters:
[image]

alisterburt, 2024-08-12:
Hi Thorsten,

Sorry but what you're doing is off the happy path that we support so you'll have to figure this one out yourself 🙂

Advice is always to pick in the tomograms reconstructed by the program you want to use for downstream averaging to avoid this kind of headache

Cheers,

Alister

closed 2024-08-12 by alisterburt