warpem/warp #196: The Parameter is Incorrect

DrJesseHansen, 2024-08-17:
I have not seen this error before, and cannot find it in any issue report or in the WARP google group.   I have been trying to refine the movie frames for a month but continuously get errors. I try removing tilt series where it crashes and new issues keep cropping up.   Most recently I turned off the "particles in CPU" option.  It ran for about half way through the data then again crashed with the error below.  

best

Jesse
```

System.IO.IOException: The parameter is incorrect.

   at System.IO.__Error.WinIOError(Int32 errorCode, String maybeFullPath)
   at System.IO.FileSystemEnumerableIterator`1.CommonInit()
   at System.IO.FileSystemEnumerableIterator`1..ctor(String path, String originalUserPath, String searchPattern, SearchOption searchOption, SearchResultHandler`1 resultHandler, Boolean checkHost)
   at System.IO.Directory.EnumerateFiles(String path, String searchPattern)
   at Warp.Movie.DiscoverParticleSuffixes(String[] fileNames) in D:\Dev\warp2\WarpLib\Movie.cs:line 702
   at Warp.Movie..ctor(String path, String[] particleFileNames) in D:\Dev\warp2\WarpLib\Movie.cs:line 766
   at Warp.TiltSeries.<>c__DisplayClass88_0.<PerformMultiParticleRefinementOneTiltMovie>b__17(Int32 i) in D:\Dev\warp2\WarpLib\TiltSeries.cs:line 6139
   at Warp.Tools.Helper.ArrayOfFunction[T](Func`2 generator, Int32 n) in D:\Dev\warp2\WarpLib\Tools\Helper.cs:line 768
   at Warp.TiltSeries.PerformMultiParticleRefinementOneTiltMovie(String workingDirectory, ProcessingOptionsMPARefine optionsMPA, Species[] allSpecies, DataSource dataSource, Movie tiltMovie, Image[] tiltMovieData, Int32 tiltID, Dictionary`2 SpeciesParticles, Dictionary`2 SpeciesParticleSubsets, Dictionary`2 SpeciesParticleIDRanges, Dictionary`2 SpeciesContainmentMasks, Dictionary`2 SpeciesRefinementSize, Dictionary`2 SpeciesRelevantRefinementSizes, Dictionary`2 SpeciesFrameWeights, Dictionary`2 SpeciesCTFSuperresFactor) in D:\Dev\warp2\WarpLib\TiltSeries.cs:line 6139
   at Warp.TiltSeries.PerformMultiParticleRefinement(String workingDirectory, ProcessingOptionsMPARefine optionsMPA, Species[] allSpecies, DataSource dataSource, Image gainRef, DefectModel defectMap, Action`1 progressCallback) in D:\Dev\warp2\WarpLib\TiltSeries.cs:line 3203
   at WarpWorker.Program.Main(String[] args) in D:\Dev\warp2\WarpWorker\Program.cs:line 294
```

dtegunov, 2024-08-22:
Could you please share the folder/file tree (just as text) for that tilt stack's directory? I haven't seen this issue before.

DrJesseHansen, 2024-08-22:
hi Dimitry,

I should have mentioned this is in M.  By tilt stack directory do you mean the warp imod folder?  That's below.  Not sure how helpful it is.
```

TS_002_ts_009.mrc/
|-- TS_002_ts_009.mrc.mrc
|-- TS_002_ts_009.mrc.rawtlt
|-- TS_002_ts_009.mrc.st
|-- TS_002_ts_009.mrc.xf
|-- aretomo_batch.sh
|-- error-aretomo_batch.sh-10041099.log
`-- output-aretomo_batch.sh-10041099.log
```

If you mean the tree for M, that's below.  I've removed some previous species (left the species which crashed) to keep things tidy, and removed some entires on "refinement_temp" for the same reason.  

```
.
├── Aug11_tightest_corrupt.population
├── Aug14_tightest.population
├── Aug17_tightest.population
├── Aug3_tightmask.population
├── Aug5_widemask.population
├── Aug7_tightestmask.population
├── global.settings
├── July23.population
├── masks
│   ├── mask_tightest_Aug7_final.mrc
│   ├── mask_tightest_Aug7.mrc
│   ├── side_only_final.mrc
│   └── binder_only.mrc
├── old
│   ├── Apr21_combined_all.population
│   ├── combined_all.population
│   └── previous.settings
├── refinement_temp
│   ├── logs
│   ├── staging
│   ├── TS_002_ts_001.mrc_fsc.mrc
│   ├── TS_002_ts_001.mrc_fscparticles.mrc
│   ├── TS_002_ts_001.mrc_fscparticles.star
│   ├── TS_002_ts_001.mrc_phaseresiduals.mrc
│   ├── worker0
│   │   └── run.out
│   ├── worker1
│   │   └── run.out
│   ├── worker2
│   │   └── run.out
│   └── worker3
│       └── run.out
└── species
├── 774ec93d
    │   ├── Aug17_angdist.mrc
    │   ├── Aug17_anisores.mrc
    │   ├── Aug17_denoised.mrc
    │   ├── Aug17_filtanisotropic.mrc
    │   ├── Aug17_filtlocal.mrc
    │   ├── Aug17_filt.mrc
    │   ├── Aug17_filtsharp.mrc
    │   ├── Aug17_fsc.star
    │   ├── Aug17_half1.mrc
    │   ├── Aug17_half2.mrc
    │   ├── Aug17_localbfac.mrc
    │   ├── Aug17_localres.mrc
    │   ├── Aug17_localres.star
    │   ├── Aug17_mask.mrc
    │   ├── Aug17_noisenet
    │   │   ├── saved_model.pb
    │   │   └── variables
    │   │       ├── variables.data-00000-of-00001
    │   │       └── variables.index
    │   ├── Aug17_particles.star
    │   ├── Aug17.species
    │   └── versions
    │       └── Aq8eF4U9
    │           ├── Aug17_angdist.mrc
    │           ├── Aug17_anisores.mrc
    │           ├── Aug17_denoised.mrc
    │           ├── Aug17_filtanisotropic.mrc
    │           ├── Aug17_filtlocal.mrc
    │           ├── Aug17_filt.mrc
    │           ├── Aug17_filtsharp.mrc
    │           ├── Aug17_fsc.star
    │           ├── Aug17_half1.mrc
    │           ├── Aug17_half2.mrc
    │           ├── Aug17_localbfac.mrc
    │           ├── Aug17_localres.mrc
    │           ├── Aug17_localres.star
    │           ├── Aug17_mask.mrc
    │           ├── Aug17_noisenet
    │           │   ├── saved_model.pb
    │           │   └── variables
    │           │       ├── variables.data-00000-of-00001
    │           │       └── variables.index
    │           ├── Aug17_particles.star
    │           └── Aug17.species
```

This probably isn't super helpful so let me know if I can help out in any other way.  In the meantime I am reprocessing all the data again in warp then will start from scratch again in M.  

Jesse Hansen

dtegunov, 2024-08-22:
Ah! I assume the tilt stacks came from desktop Warp 1.0.9? Are you still using it for processing, or did you move to WarpTools?

DrJesseHansen, 2024-08-22:
Desktop Warp 1.0.9 correct.  I am still attempting to process with desktop M using the stacks processed in Desktop warp, but running into issues at the end of the pipeline.   In parallel I'm attempting to reprocess in WarpTools.