warpem/warp #197: Improvement Suggestion: Masking pixel replacement leads to tomogram artifacts

bbarad, 2024-08-21:
We have been working on removing gold beads from tomograms in order to improve tomogram visual quality as well as denoising, segmentation, and template matching efforts. We've been masking regions prior to tomogram reconstruction (using [fidder](https://github.com/teamtomo/fidder) to generate masks) and have run into into some artifacts in the way masked regions are removed during tomogram reconstruction by warptools (and, I think, by warp):

[image]

On the left is a warptools reconstruction using masks generated by fidder, the right are reconstructions without masking, but with the averages folder manually gold-erased by fidder. The gold beads in the warptools reconstruction show up as a very clearly different structure, whereas they are fully "gone" in the manually fidder-erased case. I think generating masks but leaving the averages as is would be the preferable workflow. 

These artifacts are better than the original gold, but get enhanced by noise2noise denoising and can even cause issues with segmentation and template matching. Since fidder's noise replacement seems to work so well (and is permissively licensed), I wonder whether it would be reasonable to replace warp's mask-erase/noise replacement routine with the one used by fidder (or something similar).

dtegunov, 2024-08-22:
Hey! I think the main difference is that fidder tries to match the noise levels of the surroundings (which is what Warp used to do), whereas Warp injects less noise now. I see how that might confuse some models that are more sensitive to the noise distribution remaining constant. Let's expose it as a parameter and set the default to 100% of the estimated level (I think it's at 10% now?).

alisterburt, 2024-09-03:
closed by #202

closed 2024-09-03 by alisterburt