warpem/warp #206: warp fails to build with cuda 12  building (aiming to replace anaconda)

schloegl, 2024-08-23:
I'm trying to build warptools w/o the need for anaconda, using cuda/12.x, gcc/12 and using pypi packages only. 
The motivation for this is because anaconda is not free software, they are suing their users. 
 https://legal.anaconda.com/policies/en/?name=terms-of-service#anaconda-terms-of-service
 https://www.theregister.com/2024/08/08/anaconda_puts_the_squeeze_on/
 https://x.com/NM_Reid/status/1825997577151525338

When trying to build warp with cuda/12, it fails with this error: 

```
:~/src/warp/NativeAcceleration/build$ make 
[  1%] Building CUDA object CMakeFiles/NativeAcceleration.dir/gtom/src/Masking/IrregularSphereMask.cu.o
/nfs/scistore16/itgrp/schloegl/src/warp/NativeAcceleration/gtom/src/Masking/IrregularSphereMask.cu(19): error: texture is not a template
   texture<tfloat, 2, cudaReadModeElementType> texIrregularSphereRadius2d;
   ^

/nfs/scistore16/itgrp/schloegl/src/warp/NativeAcceleration/gtom/src/Masking/IrregularSphereMask.cu(50): error: identifier "cudaBindTexture2D" is undefined
    cudaBindTexture2D(
    ^

/nfs/scistore16/itgrp/schloegl/src/warp/NativeAcceleration/gtom/src/Masking/IrregularSphereMask.cu(67): error: identifier "cudaUnbindTexture" is undefined
    cudaUnbindTexture(texIrregularSphereRadius2d);
    ^

/nfs/scistore16/itgrp/schloegl/src/warp/NativeAcceleration/gtom/src/Masking/IrregularSphereMask.cu(109): error: no instance of overloaded function "tex2D" matches the argument list
            argument types are: (<error-type>, float, float)
     tfloat radius = tex2D(texIrregularSphereRadius2d, phi, theta);
                     ^

4 errors detected in the compilation of "/nfs/scistore16/itgrp/schloegl/src/warp/NativeAcceleration/gtom/src/Masking/IrregularSphereMask.cu".
make[2]: *** [CMakeFiles/NativeAcceleration.dir/build.make:1122: CMakeFiles/NativeAcceleration.dir/gtom/src/Masking/IrregularSphereMask.cu.o] Error 2
make[1]: *** [CMakeFiles/Makefile2:83: CMakeFiles/NativeAcceleration.dir/all] Error 2
make: *** [Makefile:91: all] Error 2

```

Do you have any suggestion how to get around this ?

alisterburt, 2024-08-23:
Hi @schloegl - I understand the recent issues with Anaconda. If you are using e.g. the mambaforge distribution of conda (the package manager) and none of the packages from Anaconda (the company) then everything is fine. None of our dependencies come from the anaconda defaults channel

alisterburt, 2024-08-23:
also: try to build with CUDA 11.8 - we haven't updated for building against CUDA 12 yet

closed 2024-08-23 by alisterburt

schloegl, 2024-08-23:
When I tried this, gcc 14.x was pulled in and that is incompatibe with cuda 11.x. cuda/11.x seems to require gcc<12.

alisterburt, 2024-08-23:
@schloegl sorry but we distribute binaries as part of the conda package and getting this working was nontrivial - we are happy to support users who want to build using the conda dependencies but we can't support everyones build system...

alisterburt, 2024-08-23:
https://github.com/warpem/warp/blob/cd1315f22b3f67b09800cacabe241698af569c8b/conda-recipe/meta.yaml#L20-L44

and logs from a successful recent build with a list of all packages/versions
https://github.com/warpem/warp/actions/runs/10396987944/job/28791942629#step:5:280

schloegl, 2024-08-24:
I was not able to setup the build environment through conda-forge, only. (I tried miniforge3 on Debian12).  However, I was able to setup the build environment using a pip venv torch=2.0.1 mkl, , install cuda 11.8, cudnn, 8.7, dotnet/8 compile fftw, tiff, cmake. It would still not install with gcc/g++ 12 as the default compiler. 
Therefore, I had to compiling gcc/g++ 10.5, from source and getting rid of libstdc++ incompatibility. 
Finally that was working for me. It clumsy but it does the job.

alisterburt, 2024-08-24:
The dependencies do not come from conda-forge only, the list of channels is visible in the installation command

schloegl, 2024-08-26:
Yes, I learned through the use of miniforge3, that conda-forge is not sufficient. One point of this ticket is avoiding anaconda and its limiting "terms-of-services".

alisterburt, 2024-08-26:
@schloegl I repeat my earlier message - our build depends on exactly zero packages from Anaconda the company.

> Hi @schloegl - I understand the recent issues with Anaconda. If you are using e.g. the mambaforge distribution of conda (the package manager) and none of the packages from Anaconda (the company) then everything is fine. None of our dependencies come from the anaconda defaults channel

schloegl, 2024-08-26:
This page https://github.com/conda-forge/miniforge says "Mambaforge (Deprecated as of July 2024) .... and configuration of Mambaforge and Miniforge3 are now identical" . I used Miniforge 24.3.0.