warpem/warp #215: threshold picks edge case failure

alisterburt, 2024-08-23:
When we have TS names which are substrings of other TS names, the matching with a wildcard matches multiple TS and throws an exception :-(

shahpnmlab, 2024-08-29:
Yes! I have come across this issue and spent quite some time thinking about it… the solution that appeals to me is [Perfect Hash Function](https://en.m.wikipedia.org/wiki/Perfect_hash_function).  Try this at your end -

```
import mmh3

class PerfectHash:
    def __init__(self, keys):
        self.keys = list(keys)
        self.size = len(self.keys)
        self.hash_table = [None] * self.size
        self.find_perfect_hash()

    def find_perfect_hash(self):
        collision = True
        seed = 0
        while collision:
            collision = False
            self.hash_table = [None] * self.size
            for key in self.keys:
                h = mmh3.hash(key, seed) % self.size
                if self.hash_table[h] is None:
                    self.hash_table[h] = key
                else:
                    collision = True
                    seed += 1
                    break
        self.seed = seed

    def get(self, key):
        h = mmh3.hash(key, self.seed) % self.size
        return self.hash_table[h]

def create_filename_matcher(filenames):
    # Create a dictionary of base names (without extension) to full filenames
    base_to_full = {name.rsplit('.', 1)[0]: name for name in filenames}
    
    # Create perfect hash for base names
    perfect_hash = PerfectHash(base_to_full.keys())
    
    def matcher(query):
        # Remove extension from query if present
        query = query.rsplit('.', 1)[0]
        
        # Try exact match first
        exact_match = perfect_hash.get(query)
        if exact_match and exact_match.startswith(query):
            return base_to_full[exact_match]
        
        # If no exact match, try partial match
        for base_name in base_to_full.keys():
            if base_name.startswith(query):
                return base_to_full[base_name]
        
        return None
    
    return matcher

# Test the function
tomogram_names_in_folder = [
    'TS_01_8.00Apx.mrc', 'TS_11_8.00Apx.mrc', 'TS_023_8.00Apx.mrc', 'TS_32_8.00Apx.mrc', 
    'TS_048_8.00Apx.mrc', 'lma2019-08-21-1.mrc', 'lma2019-08-21-11.mrc', 
    '20200427_ecoli_multi_tomo_002_s2_10.00Apx.mrc', 'TS_037.mrc', 
    'raw1Position_10_10.00Apx.mrc', 'grid6201_1_lam2_ts_006_15.00Apx.mrc', 
    'Tomo_G2_02_15.00Apx.mrc', 'lam2_poly2_ts_008.mrc_15.00Apx.mrc', 'TS3_ts_012_15.00Apx.mrc'
]

match_filename = create_filename_matcher(tomogram_names_in_folder)

test_cases = [
    'TS_1', 'TS_11', 'TS_23', 'TS_32', 'TS_48', 'lma2019-08-21-1', 'lma2019-08-21-11', 
    '20200427_ecoli_multi_tomo_002_s2', '20200427_ecoli_multi_tomo_002_s2_10.00Apx', 
    'TS_037', 'TS_037_10.00Apx', 'raw1Position_10', 'raw1Position_10_10.00Apx', 
    'grid6201_1_lam2_ts_006', 'grid6201_1_lam2_ts_006_15.00Apx', 'Tomo_G2_02', 
    'Tomo_G2_02_15.00Apx', 'lam2_poly2_ts_008', 'lam2_poly2_ts_008_15.00Apx', 
    'TS3_ts_012', 'TS3_ts_012_15.00Apx', 'Nonexistent_tomo'
]

for case in test_cases:
    match = match_filename(case)
    print(f"Base name: {case}")
    print(f"Matched with: {match}")
    print()
```

I am not claiming that this is a perfect solution, but I like it more than constructing regexps.

alisterburt, 2024-09-09:
this is actually handled pretty well already by https://github.com/warpem/warp/blob/0df8e56403f23dcfb4f712738e198040e74de73c/WarpTools/Commands/General/ThresholdPicks.cs#L84-L93

closed 2024-09-09 by alisterburt