warpem/warp #216: The semaphore timeout period has expired.

DrJesseHansen, 2024-08-24:
I think I am discovering all the new errors nobody has seen before! hah.

Processing in desktop Warp 1.0.9.   Was getting errors previously so I decided to start completely over and reprocessed the data in Warp.  Then Setup a new M session and ran poses, image warp, CTF, but when I tried adding in tilt angles and volume warp (same time) it crashed with the error below.   

I am running on a remote machine with 4x 2080Ti cards.  There were two different worker errors, clearly showing a memory issue, then one I've never seen before.

first, on worker 0:
```
System.Exception: cudaErrorMemoryAllocation
   at Warp.GPU.CheckGPUExceptions() in D:\Dev\warp2\WarpLib\GPU.cs:line 1183
   at WarpWorker.Program.Main(String[] args) in D:\Dev\warp2\WarpWorker\Program.cs:line 322

```
then one hour later on worker 3:

System.IO.IOException: The semaphore timeout period has expired.

   at System.IO.__Error.WinIOError(Int32 errorCode, String maybeFullPath)
   at System.IO.FileStream.Init(String path, FileMode mode, FileAccess access, Int32 rights, Boolean useRights, FileShare share, Int32 bufferSize, FileOptions options, SECURITY_ATTRIBUTES secAttrs, String msgPath, Boolean bFromProxy, Boolean useLongPath, Boolean checkHost)
   at System.IO.FileStream..ctor(String path, FileMode mode, FileAccess access, FileShare share)
   at Warp.TiltSeries.SaveMeta() in D:\Dev\warp2\WarpLib\TiltSeries.cs:line 8223
   at WarpWorker.Program.Main(String[] args) in D:\Dev\warp2\WarpWorker\Program.cs:line 296

I'll try re-running with only stage angles and I'll turn off "use CPU memory for particles". 

J

alisterburt, 2024-08-26:
Hi @DrJesseHansen - just to confirm, this is an older windows build, right?

DrJesseHansen, 2024-08-27:
this is windows warp yes, but the latest version of it.   I think I am about ready to transition fully to Warp 2.X though.  Btw this error is also occasionally happening with RELION 5 so I'm wondering if it's something to do with our cluster (or my data).

dtegunov, 2024-08-27:
This looks like the GPU is just running out of memory, no? I think everything else just cascades from there.

DrJesseHansen, 2024-08-28:
That could make sense. As mentioned above, the initial worker error was memory related.  I'll ditch desktop Warp for now (giving me more flexibility for nodes I can run it on) and we'll see if this ever arises again.  Thanks!

alisterburt, 2024-09-02:
closing as likely an out of memory error rather than a problem with Warp itself

closed 2024-09-02 by alisterburt