warpem/warp #217: Add multi-node documentation/script

alisterburt, 2024-08-26:


sagarbiophysics, 2024-10-29:
Hi Alister, 

I was just trying to look into the possibility to run Warptools on multinode setup. Is it possible? 

Cheers,
Sagar

alisterburt, 2024-10-29:
Hey @sagarbiophysics ! Yes but lacks a convenient API to launch everything

The general gist is:
- launch worker processes on `hostname:port` with `WarpWorker --device <gpu_id> --port <port>`
- launch your warptools command with `--workers` as a space separated list of `hostname:port`

sagarbiophysics, 2024-11-05:
@alisterburt 

Thanks a lot! Initial test seems to work but I need to figure out SLURM setup so that I can deploy on multi node multi GPU HPC. 

I will share my slurm submit files once I am done testing and things work as expected.

alisterburt, 2024-11-06:
@sagarbiophysics hero, ta!