warpem/warp #218: Can't copy a file to itself error

DcShepherd, 2024-08-27:
Hello Alister,

I seem to be having some problems with aligning my tilt series using the Imod wrappers e.g. ``` WarpTools ts_etomo_patches ```

I get the following error

```
Failed to import alignments:
Could not find Leg_1.xf
1/1, 00:00 remaining                                                                                                           
Finished processing in 00:01:21
Saying goodbye to all workers... Done 
```
Checking the etomo err log I see the message 

```
Warning:Can't copy a file to itself:source:/media/doulin/Secondary_drive/warp_test/warp_tiltseries/tiltstack/Leg_1/batchDirective.adoc,destination:/media/doulin/Secondary_drive/warp_test/warp_tiltseries/tiltstack/Leg_1/batchDirective.adoc
java.lang.Exception: Stack trace
	at java.base/java.lang.Thread.dumpStack(Thread.java:1383)
	at etomo.util.Utilities.copyFile(Utilities.java:1574)
	at etomo.util.Utilities.copyFile(Utilities.java:1555)
	at etomo.ApplicationManager.copyDirectiveFiles(ApplicationManager.java:953)
	at etomo.ApplicationManager.doneSetupDialog(ApplicationManager.java:506)
	at etomo.ui.SetupReconUIHarness.doDirectiveAutomation(SetupReconUIHarness.java:132)
	at etomo.ui.SetupReconUIHarness.doAutomation(SetupReconUIHarness.java:98)
	at etomo.ApplicationManager.doAutomation(ApplicationManager.java:307)
	at etomo.EtomoDirector.doAutomation(EtomoDirector.java:261)
	at etomo.EtomoDirector.openTomogramAndDoAutomation(EtomoDirector.java:920)
	at etomo.ui.FrontPageUIHarness.reconActionForAutomation(FrontPageUIHarness.java:44)
	at etomo.FrontPageManager.doAutomation(FrontPageManager.java:84)
	at etomo.EtomoDirector.doAutomation(EtomoDirector.java:239)
	at etomo.EtomoDirector.setup(EtomoDirector.java:190)
	at etomo.EtomoDirector.main(EtomoDirector.java:124)
java.lang.InterruptedException: sleep interrupted
	at java.base/java.lang.Thread.sleep(Native Method)
	at etomo.EtomoDirector$UtilityThread.run(EtomoDirector.java:1569)
	at java.base/java.lang.Thread.run(Thread.java:829)
exitValue:0
```

I am not really sure if I messed up to settings file. Any input would be greatly appreciated

alisterburt, 2024-08-27:
Hi @DcShepherd - this is weird, it's a Java error which means it's internal to IMOD/etomo and we can't do anything, could you try reinstalling IMOD or using a different version?

DcShepherd, 2024-08-28:
Hi @alisterburt ,
I think I sorted it out, the error is there in the err log file regardless of it the alignment works or not (at least for me). The real problem was that I forgot to at the ```--dont_invert`` flag during the import step . I assume this causes the imod alignment to fail and not produce a .xf.

I am a bit confused on the the geometry of my tomograms but at least I can further along in the pipeline

closed 2024-08-28 by DcShepherd

alisterburt, 2024-08-28:
Huh, thanks for reporting back. I can't think of any good reason that the presence/absence of --do t_invert would change how the alignment goes... maybe it does matter when fiducials are tracked?