warpem/warp #220: ts_template_match Nullable object must have value (--tilt_range)

asarnow, 2024-08-28:
`ts_template_match` produces a "nullable object must have value" error unless `--tilt_range` is given. The value can't be negative so 90 must be given instead of -1 for the default behavior.

Something about [TemplateMatchTiltseries:175](https://github.com/warpem/warp/blob/cd1315f22b3f67b09800cacabe241698af569c8b/WarpTools/Commands/Tiltseries/TemplateMatchTiltseries.cs#L175).

`OptionsMatch.TiltRange = CLI.TiltRange != null ? (decimal)CLI.TiltRange.Value : -1;`

alisterburt, 2024-08-28:
Thanks for the report @asarnow - will check what's up once I'm back from travelling! 🙂

alisterburt, 2024-09-02:
@asarnow I can't reproduce either on a build of main at [15f4ab3](https://github.com/warpem/warp/commit/15f4ab3ae5963c932b0f9f97d4da60daa175f781) or `warp=2.0.0dev25` from conda - will close for now but do reopen if you think this is still a problem...

closed 2024-09-02 by alisterburt