warpem/warp #221: WarpTools: Picking and exporting particles from averaged micrographs

mjaremko, 2024-08-29:
Is there a better way to pick and export particles from already averaged micrographs?

I was successfully able to picking and generate a star file with this workflow:

WarpTools create_settings --output warp_frameseries.settings --folder_data . --extension "*.mrc" --angpix 0.746
mkdir average
ln -sr *.mrc average
WarpTools fs_boxnet_infer --settings warp_frameseries.settings --model ../BoxNet_20220403_191253.pt --diameter 150 --threshold 0.2 --device_list 6 7
WarpTools fs_export_particles --settings warp_frameseries.settings --input matching --output good_particles.star --box 4 --diameter 150 --angpix_coords 0.746 --device_list 6 7

But the 'WarpTools fs_export_particles' command generates the star file with coordinates and 'MicrographName' field, but the ImageName column is empty and now particle stacks are generated. The job does produce a 'Failed to process...' error but there is nothing in the log. 

Also, if I process from MotionCorrection>CTFestimation the 'WarpTools fs_export_particles' produces the particle stacks and complete star file.

alisterburt, 2024-08-29:
Hi @mjaremko ! Thanks for thoroughly looking at this.

The workflow you're trying to do is not directly supported/expected so it's not surprising that the export fails but it would be nice to make it work - could you share a couple of files with working and not working scripts? This will make fixing the outputs much easier 🙂

mjaremko, 2024-08-29:
Hi Alister,

I used a couple .eer movies from EMPIAR-10424. I can't upload here (files are too big), but I can drop them in a link if you have one. Or you can download a couple from EMPIAR. Let me know.

Here are the commands:

#EMPIAR 10424 test full
WarpTools create_settings --output warp_frameseries.settings --folder_data Movies --folder_processing warp_frameseries --extension "*.eer" --angpix 0.457
WarpTools fs_motion_and_ctf --settings warp_frameseries.settings c_defocus_min --out_averages --out_average_halves
WarpTools fs_boxnet_infer --settings warp_frameseries.settings --model BoxNet_20220403_191253.pt --diameter 150 --threshold 0.2 --device_list 6 7
WarpTools fs_export_particles --settings warp_frameseries.settings --input warp_frameseries/matching/ --output good_particles.star --box 4 --diameter 150 --angpix_coords 0.746 --device_list 6 7

#EMPIAR 10424 test from micrographs (no particle stacks)
WarpTools create_settings --output warp_micrographs.settings --folder_data micrographs --folder_processing warp_micrographs --extension "*.mrc" --angpix 0.457
mkdir -p warp_micrographs/average
ln -sr micrographs/*.mrc warp_micrographs/average
WarpTools fs_boxnet_infer --settings warp_micrographs.settings --model BoxNet_20220403_191253.pt --diameter 150 --threshold 0.2 --device_list 6 7
WarpTools fs_export_particles --settings warp_micrographs.settings --input warp_micrographs/matching/ --output good_particles.star --box 4 --diameter 150 --angpix_coords 0.746 --device_list 6 7

mjaremko, 2024-08-29:
https://www.ebi.ac.uk/empiar/EMPIAR-10424/

alisterburt, 2024-08-30:
@mjaremko thanks, could you link to the boxnet weights you are using?

mjaremko, 2024-08-30:
Oh right:) You can find it in this zip file for Warp 1.1 beta:
https://seafile.unistra.fr/d/44157c1e129544bb9c43/
located here in zipped folder:
./Warp_110_beta1/boxnet3models/BoxNet_20220403_191253.pt

mjaremko, 2024-09-16:
Hi Alister,
Were you able to access the boxnet weights? Any success with the fixes? Thanks!

mjaremko, 2024-09-17:
Thanks for the update Alister, no worries.

inetnils, 2025-06-04:
Dear Alistair,

I am trying to attempt more or less the same doing particle picking for SPA with warptools from averages obtained from Relion.

I did:

`WarpTools create_settings -o warpinit.settings --folder_processing ./Warp_processing --folder_data ./Micrographs/ --extension yourfilenames*.tif --angpix 0.zzz --gain_path Micrographs/Your_CountRef.mrc --exposure xx.yy`
Then create a symbolic link` ln -s MotionCorr/jobxxx/Micrographs/ Warp_processing/average
`
`WarpTools fs_boxnet_infer --settings warpinit.settings --model path_to_model.pt --diameter xxx --threshold y.yy --distance zz --suffix_star manualpick`

**As a model, I used the model that can be found in the windows Warp Gui and changed the file ending to pb.** I did not find any documentation on where to obtain the model from and this did not throw any errors. However, I am doubting that this is the correct way. Is there a way to get the BoxNet2Mask_20180918 model?

In Relion folder:
rm ManualPick/jobxyz/Micrographs

ln -s yourfolder/Warp_processing/matching/ ManualPick/jobxyz/Micrographs to link the warp picked micrographs into the folder where relion can view them.

In Relion, if you start the Manual pick process, you can view the picks (and thus can proceed in relion I guess).

My problem is, however, that the picks do not make sense. I compared the picks obtained from the Windows Gui Warp( where my protein is picked very well) with the results from warptools (where all the picks are in the void) so I am wondering if the there is an issue with reusing the model from the Gui? Or is there an issue with particle location conventions?

Thank you in advance for your help,
Nils