warpem/warp #231: `inf` in motion corrected average

alisterburt, 2024-09-02:
from the EMBO workshop, a user had some files which don't contain NaNs. Motion correction ran successfully. Subsequent CTF estimation failed.

Input file:

```txt
(warp_build) [burta2@sc1ns017is13 hamad]$ header frames/12sep17_3_367_0.0_Sep12.mrc 

 RO image file on unit   1 : frames/12sep17_3_367_0.0_Sep12.mrc     Size=     166864 K

 Number of columns, rows, sections .....    3710    3838       6
 Map mode ..............................    6   (unsigned 16-bit integer)  
 Start cols, rows, sects, grid x,y,z ...    0     0     0    3710   3838      6
 Pixel spacing (Angstroms)..............   2.200      2.200      2.200    
 Cell angles ...........................   90.000   90.000   90.000
 Fast, medium, slow axes ...............    X    Y    Z
 Origin on x,y,z .......................    0.000       0.000       0.000    
 Minimum density .......................   0.0000    
 Maximum density .......................   38857.    
 Mean density ..........................   25.079    
 tilt angles (original,current) ........   0.0   0.0   0.0   0.0   0.0   0.0
 Space group,# extra bytes,idtype,lens .        0        0        0        0

     1 Titles :
SerialEMCCD: Dose frac. image, scaled by 32.00  r/f 3   12-Sep-17  06:37:48    
```

```ipython
In [1]: import numpy as np

In [2]: import mrcfile

In [3]: frames = mrcfile.read('frames/12sep17_3_367_0.0_Sep12.mrc')

In [4]: mc_avg = mrcfile.read('warp_fs/average/12sep17_3_367_0.0_Sep12.mrc')

In [5]: np.max(frames)
Out[5]: np.uint16(38857)

In [6]: np.max(mc_avg)
Out[6]: np.float16(inf)
```

alisterburt, 2024-09-03:
this is due to a few hot pixels - hot pixel removal used to be a thing in Warp but Dimitry removed it because it was hard to make robust with counting data. Recommendation would be to run [`ccderaser` from IMOD](https://bio3d.colorado.edu/imod/doc/man/ccderaser.html) before importing into warp - could also add a warning on movie export

alisterburt, 2024-09-03:
closed by #233 - not a fix but at least an explicit warning prior to subsequent failure 🤷🏼‍♂️

closed 2024-09-03 by alisterburt

dtegunov, 2024-09-03:
Switching to fp32 MRCs might be a better workaround