warpem/warp #232: star file compatibility

jowagner91, 2024-09-03:
Hey, I noticed that warp tools ts_export_particles requires a very specific structure for star files. 

If I run TM with an external tool and write the star using the star file package, partial extraction failed till I modified the header to only one: 
/n 
data_
....
If there is any additional information above the data_ (like the line normally added by the starfile python package) parsing fails (tested on warp/2.0dev23). 
I think this should either be documented somewhere on the tutorial page or the parser changed so it is a bit more flexible.

alisterburt, 2024-09-03:
huh, I'm the original author of `starfile` and I used to use files written with it in Warp all the time, wonder how this regression happened... will take a look, thanks for the report!

alisterburt, 2024-09-03:
@jowagner91 I can't reproduce your issue

With a file that looks like this:

```txt
# Created by the starfile Python package (version 0.5.8) at 04:17:44 on 03/09/2024

data_

loop_
_rlnCoordinateX #1
_rlnCoordinateY #2
_rlnCoordinateZ #3
_rlnAngleRot #4
_rlnAngleTilt #5
_rlnAnglePsi #6
_rlnMicrographName #7
_rlnAutopickFigureOfMerit #8
0.853448	0.856540	0.269231	112.499990	11.715852	22.500000	TS_11.tomostar	6.573487
...
0.163793	0.930380	0.730769	45.000000	5.850267	247.499980	TS_17.tomostar	6.868605

```

I can successfully write particle stacks with the following command:

```sh
WarpTools ts_export_particles \
--settings warp_tiltseries.settings \
--input_star test_starfile.star \
--normalized_coords \
--output_star particle_export_test/matching.star \
--output_processing particle_export_test \
--output_angpix 4 \
--box 96 \
--diameter 130 \
--relative_output_paths \
--2d 
```

Not sure what's different to yours - will close this for now but feel free to reopen if you narrow down the issue further

closed 2024-09-03 by alisterburt