warpem/warp #243: Suggestion regarding RELION 2D subtilts with no visble frames

rkjensen, 2024-09-11:
Running RELIONv5 using extracted 2D subtilts from warp, where some particles are too close to the blurred corners (i.e. leading to a _rlnTomoVisibleFrames entry = [0,0,0, ... 0]) causes a breaking error in RELION. The error message is very unhelpful and it took me some time to track down the real issue.

As these particles are not contributing any signal anyway (and are most likely false positives) it would be nice if warp could just exclude those from being extracted automatically (perhaps including some warning message, and a flag to disable this behaviour).

Example of a particle entry where this is the case is below:

data_particles

loop_
_rlnTomoName #1
_rlnTomoParticleId #2
_rlnCoordinateX #3
_rlnCoordinateY #4
_rlnCoordinateZ #5
_rlnAngleRot #6
_rlnAngleTilt #7
_rlnAnglePsi #8
_rlnTomoParticleName #9
_rlnOpticsGroup #10
_rlnImageName #11
_rlnOriginXAngst #12
_rlnOriginYAngst #13
_rlnOriginZAngst #14
_rlnTomoVisibleFrames #15
00035.tomostar	14	23.389	1809.712	92.851	-165.633	122.283	94.405	00035/14	5	/scratch/kjeldsen/warp_runs/particles/wt1/particleseries/00035/00035_3.40A_000014.mrcs	0.0	0.0	0.0	[0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0]

alisterburt, 2024-09-11:
Hi @rkjensen, sure! Sounds reasonable - is it only particles which are completely not visible you think we should get rid of? What about if only some frames are visible?

alisterburt, 2024-09-18:
@rkjensen thanks again for the report. I added an option `--max_missing_tilts` which defaults to 5 in #252 and should avoid this problem in the future. Will be available in `2.0.0dev27` once **https://github.com/warpem/warp/actions/runs/10927356218** finishes!

cc @rahelwoldeyes I think you ran into the same thing based on the RELION discussion

rahelwoldeyes, 2024-09-18:
Thank you! @DrJesseHansen @KrisJanssen might also be interested to hear about the update.

DrJesseHansen, 2024-09-18:
fantastic.   Thanks @alisterburt !  Also @rkjensen and @rahelwoldeyes

PR #252 (2024-09-18): add --max_missing_tilts option to particle series export
closes #243 - pending testing