warpem/warp #244: Unable to specify GPU ID using --device_list in WarpTools fs_motion_and_ctf

xinsheng44, 2024-09-12:
I encountered an issue when trying to specify the GPU ID for processing with the --device_list parameter in WarpTools fs_motion_and_ctf. According to the documentation, --device_list accepts a "space-separated list of GPU IDs to use for processing. Default: all GPUs in the system." However, the machine I'm using has multiple users, and GPU 0 is already occupied. When processing defaults to GPU 0, I receive an error.

I tried specifying GPU 1 using:

--device_list 1
--device_list [1]
--device_list {1}
None of these formats worked, and I still couldn't force the processing to use a specific GPU. Could you please clarify the correct syntax or provide guidance on how to properly specify the GPU ID?

Thank you!

xinsheng44, 2024-09-12:
> I encountered an issue when trying to specify the GPU ID for processing with the --device_list parameter in WarpTools fs_motion_and_ctf. According to the documentation, --device_list accepts a [...]

I thought the issue was an incompatibility between CUDA and cuFFT versions, but both are CUDA 11.8.0, as shown below:

nvcc --version
nvcc: NVIDIA (R) Cuda compiler driver
Copyright (c) 2005-2022 NVIDIA Corporation
Built on Wed_Sep_21_10:33:58_PDT_2022
Cuda compilation tools, release 11.8, V11.8.89
Build cuda_11.8.r11.8/compiler.31833905_0

conda list | grep cufft
libcufft                  10.9.0.58                     0    nvidia/label/cuda-11.8.0
libcufft-dev              10.9.0.58                     0    nvidia/label/cuda-11.8.0

I referred to the  issue [https://github.com/warpem/warp/issues/172](url) again, reinstalled a clean environment, and switched to CUDA 11.7.0, but the same error still occurs.

xinsheng44, 2024-09-12:
I just tested it, and using “--device_list 1” allows specifying the GPU ID. The error didn’t occur this time, which is really strange because I tried many times earlier and it didn’t work. Now, it suddenly works. 

To summarize, the reason for my error was that I didn’t set “--device_list”, which caused the program to automatically assign GPUs. Some of these GPUs were being used by other users, leading to insufficient memory. 

The solution is to specify the GPU ID, and it works.

alisterburt, 2024-09-12:
@xinsheng44 thanks for updating here and glad you got sorted, closing!

closed 2024-09-12 by alisterburt