warpem/warp #253: WarpTools from Scipion.

pconesa, 2024-09-20:
We are integrating WarpTools in Scipion.... and we have an issue.

we are running the command Warptools fs_motion_and_ctf xxx

It runs ok from a terminal but not from Scipion. We believe the reason is that from Scipion there is no "physical" console and threfore ConsoleWith is 0... then here Warp tries to create a string of -2 spaces:

https://github.com/warpem/warp/blob/main/WarpLib/Tools/VirtualConsole.cs#L77

Maybe there is a option we haven't found to make the process silent? 

One more question.... we are able to run the ts_ctf from Scipion this way.

1.- We start from a tilt series stack + metadata. We are able to generate the tomostar, but at this point there aren't any "frames". We do not have frames at this point. Even frames might not be available in the Scipion project cause users may have enter Scipion from an already stacked tile series.

2.- We create the averages folder for warp "unstacking our tilt series stack, to mimic warp folder structure as if they were motion-corrected by warp.... but still it looks for the frames to read the dimensions. We are placing in the frames folder that same unstack average (not frames) but seems to be enough for the process to run.  (hack) https://github.com/warpem/warp/blob/main/WarpLib/TiltSeries.cs#L354

Is this risky? Is there a way to run ts_ctf without having frames? I think will make this command more used in other scenarios where not all have been done in warp.

A 3rd question, the output of the tilt series ctf XML: We want to store the CTF per each of the tilts.... is it correct to take them from the GridCTF element? We can see some delta values not sure we need to consider.

alisterburt, 2024-09-20:
Hi @pconesa - I just pushed #254 to fix the issue you mentioned and it will be released in `2.0.0dev28` when this workflow finishes https://github.com/warpem/warp/actions/runs/10962865116

I'll follow up later about the second part of your issue 🙂

alisterburt, 2024-09-20:
> Is this risky? Is there a way to run ts_ctf without having frames? I think will make this command more used in other scenarios where not all have been done in warp.

It's not especially risky and should work fine - if you run into any issues don't hesitate to bring them up here.

I agree that it would be nice if Warp/WarpTools were more modular and could be used as standalone tools but unfortunately that's not the way the system is architected 🙁 

> A 3rd question, the output of the tilt series ctf XML: We want to store the CTF per each of the tilts.... is it correct to take them from the GridCTF element? We can see some delta values not sure we need to consider.

The `GridCTF` values (with one grid point per tilt) are indeed the correct defoci - the `GridCTFDefocusAngle`/`GridCTFDefocusDelta` values describe the astigmatism 🙂

alisterburt, 2024-09-20:
I'm going to close this issue as I think everything is answered, please open back up if something is missing!

closed 2024-09-20 by alisterburt

pconesa, 2024-09-20:
Nice! Thanks a lot

pconesa, 2024-09-23:
@alisterburt , I've seen there is a release but seems to have the wrong macro?

https://github.com/warpem/warp/releases/tag/v0.0.28

alisterburt, 2024-09-23:
@pconesa I don't understand what you mean by 'the wrong macro'

pconesa, 2024-09-24:
you seem to have released the 28th version but ended up to be 0.0.28..

I guess you wanted 2.0.28 ?

pconesa, 2024-09-24:
Versions number are name like this:  macro.minor.micro. You have released the latest version with a 0 in the macro nombre. Sorry for not being clear.

We will test 0.0.28 specifying the version for now.

dtegunov, 2024-09-26:
@alisterburt @pconesa  FYI, I've changed the ClearLastLine behavior to append \r when not in a real console window: 1665207

pconesa, 2024-09-26:
Thanks both the quick response.

We have started to call warptools from Scipion and so far is going well. The only error we found was the reported here and is no fixed.

Thanks for all, Pablo.

Scipion team