warpem/warp #255: Failed to import alignments: Could not find *.xf

xinsheng44, 2024-09-21:
When I use the command 
`WarpTools ts_etomo_patches --settings warp_tiltseries.settings --angpix 10.88 --patch_size 500 --initial_axis 177.2 --perdevice 5`,
 I encountered an error. 

I referred to [https://github.com/warpem/warp/issues/225](url) and found that I do have the tilt file, but it still cannot find the *.xf file.

[image]

alisterburt, 2024-09-21:
@xinsheng44 can you check the logs in your tilt series processing directory? It looks like IMOD failed to align this tilt series

alisterburt, 2024-09-21:
Also it's not recommended to use --perdevice with the tilt series alignment wrapper programs

xinsheng44, 2024-09-21:
> @xinsheng44 can you check the logs in your tilt series processing directory? It looks like IMOD failed to align this tilt series

[image]

Based on the logs, the error occurred during memory allocation. However, I monitored the server, and the highest memory usage was only slightly over 40GB, while the total available memory is 1TB.

xinsheng44, 2024-09-21:
> Also it's not recommended to use --perdevice with the tilt series alignment wrapper programs

Oh, I was referring to [this](https://github.com/warpem/warp/blob/main/scripts/EMPIAR-10491_5TS_e2e.sh).

alisterburt, 2024-09-21:
That shouldn't be there, sorry about that. 

Try running without per device 5 and see if that helps with your memory issue

xinsheng44, 2024-09-21:
> That shouldn't be there, sorry about that.
> 
> Try running without per device 5 and see if that helps with your memory issue

It doesn't seem to be the problem. I referred to [this tutorial](https://warpem.github.io/warp/user_guide/warptools/quick_start_warptools_tilt_series/) related to EMPIAR_10491 and successfully ran through the entire process, achieving a resolution of approximately 2.9 Å. However, with my own dataset, I encountered this error at this particular step （WarpTools ts_etomo_patches ）. The error persists whether I remove or retain the 'perdevice' option

alisterburt, 2024-09-23:
@xinsheng44 if it works correctly on the tutorial and not on your data then something is different about your data

- maybe patch tracking just doesn't work for your dataset
- maybe your initial tilt axis was wrong
- maybe your patch size is too small

Try opening up the .edf file with etomo and seeing how well the patches are tracked

I'm closing this now as this isn't a WarpTools issue, the program is running fine - the issue is somewhere between your parameters, IMOD and your data

closed 2024-09-23 by alisterburt