warpem/warp #256: skipping motion correction while using WarpTools fs_motion_and_ctf

rahelwoldeyes, 2024-09-21:
Is there a way to skip motion correction while using WarpTools fs_motion_and_ctf? I tried:

- removing m_grid
- changing m_grid=1x1x1

I get an error message: 

> System.NullReferenceException: Object reference not set to an instance of an object.
>    at Warp.Tools.MathHelper.Diff(float2[] data) in /usr/share/miniconda/envs/package-build/conda-bld/warp_1726681409458/work/WarpLib/Tools/MathHelper.cs:line 638
>    at Warp.Movie.ProcessShift(Image originalStack, ProcessingOptionsMovieMovement options) in /usr/share/miniconda/envs/package-build/conda-bld/warp_1726681409458/work/WarpLib/Movie.cs:line 2455
>    at WarpWorker.WarpWorker.EvaluateCommand(NamedSerializableObject Command) in /usr/share/miniconda/envs/package-build/conda-bld/warp_1726681409458/work/WarpWorker/WarpWorker.cs:line 277

alisterburt, 2024-09-21:
Hey @rahelwoldeyes - why not just run fs_ctf?

rahelwoldeyes, 2024-09-21:
Hi @alisterburt  I can't directly use fs_ctf because I can't import the tilts without frame series metadata. Here is the error message: 

> Found 30 MDOC files, searching for *.mdoc in /X/mdoc
Looking for frame series...
Unhandled exception. System.Exception: No frame series metadata found at /X/tilts
   at WarpTools.Commands.ImportTiltseries.Run(Object options) in /usr/share/miniconda/envs/package-build/conda-bld/warp_1726681409458/work/WarpTools/Commands/Tiltseries/ImportTiltseries.cs:line 118
   at WarpTools.WarpTools.Run(Object options) in /usr/share/miniconda/envs/package-build/conda-bld/warp_1726681409458/work/WarpTools/Program.cs:line 30
   at Warp.Tools.CommandLineParserHelper.ParseAndRun(String[] args, Func`2 run, Type[] verbs, String appName) in /usr/share/miniconda/envs/package-build/conda-bld/warp_1726681409458/work/WarpLib/Tools/CommandLineParserHelper.cs:line 26
   at WarpTools.WarpTools.Main(String[] args) in /usr/share/miniconda/envs/package-build/conda-bld/warp_1726681409458/work/WarpTools/Program.cs:line 17
   at WarpTools.WarpTools.<Main>(String[] args)
Aborted (core dumped)

Here is the workflow I am using:

> WarpTools create_settings \
> --folder_data tilts \
> --folder_processing warp_frameseries \
> --output warp_frameseries.settings \
> --extension "*.mrc" \
> --angpix 2.254 \ \
> --exposure 3.05
Running command create_settings with:
output = warp_frameseries.settings
folder_processing = warp_frameseries
folder_data = tilts
recursive = False
extension = *.mrc
angpix = 2.254
bin = null
bin_angpix = null
gain_path = null
defects_path = null
gain_flip_x = False
gain_flip_y = False
gain_transpose = False
exposure = 1
eer_ngroups = 40
eer_groupexposure = null
tomo_dimensions = null

> 
> Settings saved to /X/warp_frameseries.settings
> Now trying to find the files referenced by these settings...
> 957 files found

> WarpTools ts_import \
> --mdocs mdoc \
> --frameseries tilts \
> --tilt_exposure 3.05 \
> --min_intensity 0.3 \
> --dont_invert \
> --output tomostar
Running command ts_import with:
mdocs = mdoc
pattern = *.mdoc
frameseries = tilts
tilt_exposure = 3.05
dont_invert = True
override_axis = null
auto_zero = False
tilt_offset = null
max_tilt = 90
min_intensity = 0.3
max_mask = 1
min_ntilts = 1
output = tomostar

> Found 30 MDOC files, searching for *.mdoc in /X/mdoc
> Looking for frame series...
> Unhandled exception. System.Exception: No frame series metadata found at /X/tilts
>    at WarpTools.Commands.ImportTiltseries.Run(Object options) in /usr/share/miniconda/envs/package-build/conda-bld/warp_1726681409458/work/WarpTools/Commands/Tiltseries/ImportTiltseries.cs:line 118
>    at WarpTools.WarpTools.Run(Object options) in /usr/share/miniconda/envs/package-build/conda-bld/warp_1726681409458/work/WarpTools/Program.cs:line 30
>    at Warp.Tools.CommandLineParserHelper.ParseAndRun(String[] args, Func`2 run, Type[] verbs, String appName) in /usr/share/miniconda/envs/package-build/conda-bld/warp_1726681409458/work/WarpLib/Tools/CommandLineParserHelper.cs:line 26
>    at WarpTools.WarpTools.Main(String[] args) in /usr/share/miniconda/envs/package-build/conda-bld/warp_1726681409458/work/WarpTools/Program.cs:line 17
>    at WarpTools.WarpTools.<Main>(String[] args)
> Aborted (core dumped)

And tomostar files are needed to run ts_ctf

>  WarpTools ts_ctf \

> --settings warp_tiltseries.settings \
> --range_high 7 \
> --defocus_max 8 \
> --perdevice 2
Running command ts_ctf with:
window = 512
range_low = 30
range_high = 7
defocus_min = 0.5
defocus_max = 8
voltage = 300
cs = 2.7
amplitude = 0.07
fit_phase = False
auto_hand = 0
device_list = {  }
perdevice = 2
workers = {  }
settings = warp_tiltseries.settings
input_data = {  }
input_data_recursive = False
input_processing = null
output_processing = null
input_norawdata = False

> No alternative input specified, will use input parameters from warp_tiltseries.settings
> File search will be relative to /X/tomostar
> Unhandled exception. System.IO.DirectoryNotFoundException: Could not find a part of the path '/X/tomostar'.
>    at System.IO.Enumeration.FileSystemEnumerator`1.CreateDirectoryHandle(String path, Boolean ignoreNotFound)
>    at System.IO.Enumeration.FileSystemEnumerator`1.Init()
>    at System.IO.Enumeration.FileSystemEnumerable`1..ctor(String directory, FindTransform transform, EnumerationOptions options, Boolean isNormalized)
>    at System.IO.Enumeration.FileSystemEnumerableFactory.UserFiles(String directory, String expression, EnumerationOptions options)
>    at System.IO.Directory.InternalEnumeratePaths(String path, String searchPattern, SearchTarget searchTarget, EnumerationOptions options)
>    at WarpTools.Commands.BaseOptions.Evaluate() in /usr/share/miniconda/envs/package-build/conda-bld/warp_1726681409458/work/WarpTools/Commands/BaseOptions.cs:line 93
>    at WarpTools.Commands.DistributedOptions.Evaluate() in /usr/share/miniconda/envs/package-build/conda-bld/warp_1726681409458/work/WarpTools/Commands/DistributedOptions.cs:line 29
>    at WarpTools.Commands.CTFTiltseries.Run(Object options) in /usr/share/miniconda/envs/package-build/conda-bld/warp_1726681409458/work/WarpTools/Commands/Tiltseries/CTFTiltseries.cs:line 57
>    at WarpTools.WarpTools.Run(Object options) in /usr/share/miniconda/envs/package-build/conda-bld/warp_1726681409458/work/WarpTools/Program.cs:line 30
>    at Warp.Tools.CommandLineParserHelper.ParseAndRun(String[] args, Func`2 run, Type[] verbs, String appName) in /usr/share/miniconda/envs/package-build/conda-bld/warp_1726681409458/work/WarpLib/Tools/CommandLineParserHelper.cs:line 26
>    at WarpTools.WarpTools.Main(String[] args) in /usr/share/miniconda/envs/package-build/conda-bld/warp_1726681409458/work/WarpTools/Program.cs:line 17
>    at WarpTools.WarpTools.<Main>(String[] args)
> Aborted (core dumped)
> 
>

alisterburt, 2024-09-21:
@rahelwoldeyes it looks like your workflow is a little off, try 
create_settings (fs) -> fs_ctf -> ts_import -> create_settings (ts) -> ts_etomo/aretomo -> ts_ctf

rahelwoldeyes, 2024-09-21:
I completely overlooked the fs_motion and fsc_ctf options!! Thank you!

closed 2024-09-21 by rahelwoldeyes

alisterburt, 2024-09-22:
☺️ glad to help, enjoy the rest of your weekend!