warpem/warp #258: What is version v0.0.28?

jjv5, 2024-09-25:
This does not make sense. Is it a typo? It is confusing.

james-vincent, 2024-09-25:
Installing warp by conda with the warpem channel results in binaries that report version 2.0.0 still:

$ ./MCore --version
MCore 2.0.0+99c272c93f862eafe26594ccc5f4042eb58e7449

This is version installed:
warp                                  0.0.28        py311_0                       371.3 MiB  conda  warp-0.0.28-py311_0.tar.bz2

alisterburt, 2024-09-25:
sorry about this - I pushed the wrong tag (cc @pconesa who also brought this up in #253 )

alisterburt, 2024-09-25:
`2.0.0dev28` deploying now https://github.com/warpem/warp/actions/runs/11037280474

closed 2024-09-25 by alisterburt

james-vincent, 2024-09-25:
Thanks for the quick response. Is the 2.0.0dev28 build available in warpem channel also?

alisterburt, 2024-09-25:
@james-vincent it will be once that deployment workflow finishes :)

jjv5, 2024-09-26:
Thanks again for the quick work. Once installed, how can I tell if I have version dev28? The version command only outputs 2.0.0. Is the string 2.0.0dev28 embedded in the code anywhere? Since these seem like pretty common and ongoing updates it would be nice to let users be able to see what version they have.

alisterburt, 2024-09-26:
'conda list warp' - I'll add a note to the docs