warpem/warp #260: Changelog for new versions?

james-vincent, 2024-09-27:
Is there a changelog for new releases?  If not, it would be very helpful to have one. Users are unable to tell if they should upgrade to the latest version or not.

james-vincent, 2024-09-27:
Specifically what has changed from dev11 to dev28?

alisterburt, 2024-09-27:
We don't maintain one - the solution for right now would be to check the dates of the tags for each version and check the commit messages between those dates
- https://github.com/warpem/warp/tags
- https://github.com/warpem/warp/commits/main/

Mostly a bunch of bugs have been fixed and a few convenience APIs on existing tools

alisterburt, 2024-09-27:
dev tags indicate very minor updates, users can and should upgrade without worrying - once we're out of pre-release we will follow https://semver.org/

closed 2024-09-27 by alisterburt