warpem/warp #261: repeated core dump after EstimateWeight

chowdermin, 2024-09-27:
I ran warp/2.0.0dev23-foss-2022a-CUDA-11.8.0 for M refinement (imagewarp, volumewarp, defocus, stageanlge, and zernike). I did Estimateweights, but then I started to have core_dumps when I tried to refine again. I replaced all .xml with those before Estimateweights and ran it0, but still it ran into coredump. I wonder how I got this issue, or how shall I modify to get it back to work? All jobs were run on L40s with at least 4 GPUs. I attached slurm.out for your reference. Thank you!
[slurm.gpu42.12681768.txt](https://github.com/user-attachments/files/17169091/slurm.gpu42.12681768.txt)

alisterburt, 2024-09-27:
Hi @chowdermin - I'm pasting the error below for ease of coming back to it

Can you please try the latest build `2.0.0dev28` and confirm that you're still seeing the same issue?

I'm not sure what would be causing this - reverting the tilt series xml files is good but it looks like you also need to revert to an earlier version of your species to get back to a working state.

```txt
Commiting changes in 20240603...Unhandled exception. Unhandled exception.Unhandled exception.Unhandled exception.   Unhandled exception. Unhandled exception. Unhandled exception. Unhandled exception. Unhandled exception. Unhandled exception. Unhandled exception. Unhandled exception. System.NotImplementedException: The method or operation is not implemented.
   at MCore.MCore.WorkerDied(Object sender, EventArgs e)
   at Warp.WorkerWrapper.ReportDeath()
   at Warp.WorkerWrapper.<StartHeartbeat>b__14_0()System.NotImplementedException: The method or operation is not implemented.
   at MCore.MCore.WorkerDied(Object sender, EventArgs e)
   at Warp.WorkerWrapper.ReportDeath()
   at Warp.WorkerWrapper.<StartHeartbeat>b__14_0()System.NotImplementedException: The method or operation is not implemented.
   at MCore.MCore.WorkerDied(Object sender, EventArgs e)
   at Warp.WorkerWrapper.ReportDeath()
   at Warp.WorkerWrapper.<StartHeartbeat>b__14_0()
System.NotImplementedException: The method or operation is not implemented.
   at MCore.MCore.WorkerDied(Object sender, EventArgs e)
   at Warp.WorkerWrapper.ReportDeath()
   at Warp.WorkerWrapper.<StartHeartbeat>b__14_0()
System.NotImplementedException: The method or operation is not implemented.
   at MCore.MCore.WorkerDied(Object sender, EventArgs e)
   at Warp.WorkerWrapper.ReportDeath()
   at Warp.WorkerWrapper.<StartHeartbeat>b__14_0()
System.NotImplementedException: The method or operation is not implemented.
   at MCore.MCore.WorkerDied(Object sender, EventArgs e)
   at Warp.WorkerWrapper.ReportDeath()
   at Warp.WorkerWrapper.<StartHeartbeat>b__14_0()
System.NotImplementedException: The method or operation is not implemented.
   at MCore.MCore.WorkerDied(Object sender, EventArgs e)
   at Warp.WorkerWrapper.ReportDeath()
   at Warp.WorkerWrapper.<StartHeartbeat>b__14_0()
System.NotImplementedException: The method or operation is not implemented.
   at MCore.MCore.WorkerDied(Object sender, EventArgs e)
   at Warp.WorkerWrapper.ReportDeath()
   at Warp.WorkerWrapper.<StartHeartbeat>b__14_0()

System.NotImplementedException: The method or operation is not implemented.
   at MCore.MCore.WorkerDied(Object sender, EventArgs e)
   at Warp.WorkerWrapper.ReportDeath()
   at Warp.WorkerWrapper.<StartHeartbeat>b__14_0()
System.NotImplementedException: The method or operation is not implemented.
   at MCore.MCore.WorkerDied(Object sender, EventArgs e)
   at Warp.WorkerWrapper.ReportDeath()
   at Warp.WorkerWrapper.<StartHeartbeat>b__14_0()
System.NotImplementedException: The method or operation is not implemented.
   at MCore.MCore.WorkerDied(Object sender, EventArgs e)
   at Warp.WorkerWrapper.ReportDeath()
   at Warp.WorkerWrapper.<StartHeartbeat>b__14_0()

/var/spool/slurm/job12681768/slurm_script: line 44: 914611 Aborted                 (core dumped)
```

alisterburt, 2024-09-27:
Could you also try running with a single GPU and seeing if the same issue comes up?

chowdermin, 2024-09-29:
Hi Alister,
I tried several things as you suggested:
- switch to warp/2.0.0dev28-foss-2022a-CUDA-11.8.0 -> still core dump as mentioned above
- switch back to warp/2.0.0dev23-foss-2022a-CUDA-11.8.0, and change to run with a single GPU -> it works. below is the GPU settings:
#SBATCH -N 1
#SBATCH --time 1-00:00:00
#SBATCH -o ./slurm.%N.%j.out
#SBAtCH -e ./slurm.%N.%j.err
#SBATCH --job-name=MRun
#SBATCH -p gpu-el8
#SBATCH -C genoa,gpu=L40s
#SBATCH -n 32
#SBATCH -G 1
#SBATCH --mem-per-gpu 193412 

- rerun EstimateWeights (EstimateWeights --source 20240603  --population m5/nuc_clean.population --resolve_frames) and do it0 again with 1GPU -> it works
- refine with 1GPU (MCore --population m5/nuc_clean.population --refine_imagewarp 5x5 --refine_particles) -> core dump again 
- refine with 4GPUs without changing .xml -> core dum again
- refine with 4GPUs after changing .xml -> it works

How is it related to the GPU number? How shall I choose how much GPU to use? Sometimes it looks to me that core dump happen stochastically.

alisterburt, 2024-09-30:
@chowdermin thanks for reporting back - it looks like there is a problem with refining after running `EstimateWeights` which may (or may not...) be multi-GPU specific. I can't promise a quick fix as we're (unfortunately) having to prioritise other things right now

chowdermin, 2024-09-30:
I see. Thank you for looking into it!

alisterburt, 2024-10-03:
tracking in #266

closed 2024-10-03 by alisterburt