warpem/warp #262: Tomogram denoising OOM (RAM not GPU) and CTF question

DcShepherd, 2024-09-30:
Hello Everyone,

I have quite a large dataset that I am trying to process, but I seem to be hitting some limits.

I have a dataset of around 500 tomograms that I would like to denoise, as I can't clearly see my particles in the reconstructed tomograms. I started using Noise2Map and noticed that it seemed to be loading each tomogram (including odd/even splits?) into RAM, which quickly filled up all the memory on my workstation and led to an out-of-memory (OOM) error. I then tried running it on our cluster with an H100 GPU and 560 GB of RAM. This configuration loaded everything into memory and trained the model. However, while denoising the tomograms, it ran out of memory again.

1. Is Warp supposed to keep everything in RAM like this, or might I be running the protocol incorrectly?

2. Since I can run everything at once I have been trying to denoise only subsets of the dataset using wildcard expressions, but I can't seem to get it to work and I get the following error:

```
Preparing data:
Unhandled exception. System.IO.DirectoryNotFoundException: Could not find a part of the path '/media/ghosal/HDD2/Warp_all_lester/warp_tiltseries/reconstruction/even/Lester__20240531_10_15.00Apx.mrc'.
   at System.IO.Enumeration.FileSystemEnumerator`1.CreateDirectoryHandle(String path, Boolean ignoreNotFound)
   at System.IO.Enumeration.FileSystemEnumerator`1.Init()
   at System.IO.Enumeration.FileSystemEnumerable`1..ctor(String directory, FindTransform transform, EnumerationOptions options, Boolean isNormalized)
   at System.IO.Enumeration.FileSystemEnumerableFactory.UserFiles(String directory, String expression, EnumerationOptions options)
   at System.IO.Directory.InternalEnumeratePaths(String path, String searchPattern, SearchTarget searchTarget, EnumerationOptions options)
   at Noise2Map.Noise2Map.Main(String[] args) in /usr/share/miniconda/envs/package-build/conda-bld/warp_1723683627331/work/Noise2Map/Noise2Map.cs:line 153
```

Is there a specific way to do this, or should I subset the data into separate  e.g. buckets of 50 and process each bucket one at a time?

3. Does pointing the Noise2Map program to the 3dctfs of the tomogram provide any benefit in terms of particle visualization?

alisterburt, 2024-09-30:
Hi @DcShepherd 

Upfront I'll say that this is unclear and should be better documented, appreciate that you're wrangling the program :-)

> Is Warp supposed to keep everything in RAM like this, or might I be running the protocol incorrectly?

It's worth separating training time from inference time when discussing: at training time tomograms are loaded into RAM to speed up particle cropping during training, this won't work well if you have 500+ tomograms in a dataset and working with subsets is the recommended workaround.

> Is there a specific way to do this, or should I subset the data into separate e.g. buckets of 50 and process each bucket one at a time?

It's unclear from your message exactly how you were trying to process subsets, was it by using `--input_data`? If so you need to point it at the tomostar files (e.g. `tomostar/TS_*.tomostar`). For denoising specifically I've had luck in the past just putting the subset of tomograms I want to denoise for both training and inference in a new folder. It sounds like you don't know which tomograms to denoise so your strategy sounds reasonable.

> Does pointing the Noise2Map program to the 3dctfs of the tomogram provide any benefit in terms of particle visualization?

These 3D CTFs will be required when we have a denoiser which is also capable of CTF deconvolution but this is not yet the case. You can safely ignore their existence 🙂

alisterburt, 2024-09-30:
have opened #263 to track and will close this now

closed 2024-09-30 by alisterburt

DcShepherd, 2024-10-01:
Hi @alisterburt 

Thanks for the info and quick reply. 

> At training time tomograms are loaded into RAM to speed up particle cropping during training

It was my bad assumption about the memory allocations. I only realized why it makes sense as no subtomos are extracted like in IsoNet or CryoCare.

>It's unclear from your message exactly how you were trying to process subsets

Sorry for being unclear I was trying to use wild cards in  the --observation1, --observation1  and --observation_combined parameters in a script I was making. I assume it did't work due to my less than stelar scripting skills. I did this because I didn't see an option in the Noise2Map parameters to point it at a specific ```.tomostar``` file.

I made a new script to make batch folders and then loop through the folders and denoise the tomograms until all are finished. Once I did that everything seems to be  working as intended.