warpem/warp #264: Is it possible to use the ts_reconstruct having a stack and a xf file?

pconesa, 2024-10-01:
hi, we manege to have the ctf estimation for the tilt series working.

Now we are with the tomogram reconstruct.

So, in scipion have already our data (image stack) and metadata (tilt angles, tilt axis angle, ... tilt alignment information ..).

We can generate xf files without problems but we do not see how can we pass this information to the ts_reconstruct command.
We also generate the settings files, and all thing that have worked to ctf but the alignment seems to be missing fo the reconstruct case.

We generate the .tomostart files and we tried adding there the _wrpAxisOffsetX, _wrpAxisOffsetY hoping to see those values in the {TS}.xml the tomo reconstruct produces but have no exit.

Is it possible?

pconesa, 2024-10-01:
Update: we had an error in the code. Seems that adding the right values in the tomostar, then the XML ends up with those values?

pconesa, 2024-10-01:
Update 2:

We are not sure how to get exact values from xf to warp for shifts and angle (alignment information).

https://github.com/warpem/warp/blob/8939b18a9d2025cbf4aa71a61107b1623bfcde7a/Warp/Controls/TaskDialogs/Tomo/DialogTomoImportImod.xaml.cs#L297

If I'm correct, the shift should remain the same but with the sign changed? But this is not what we find in the xml.

pconesa, 2024-10-02:
Ok, we were not doing matix multiplication properly.. We now get the right values.

alisterburt, 2024-10-02:
@pconesa looks like you got sorted here, let me know if any more questions come up!

pconesa, 2024-10-04:
We are currently experiencing an issue:

Initial setup:

We generate the title series setting using Warptools
We generate "programmatically" the tomostar file (with alignment fields)
We call ts_reconstruct

We can see that TS_01.xml is generated with the right alignment fields. but I'm now realizing that we are not passing the CTF values.

Warp ts_reconstruct command gets stuck processing. 

I guess CTF is mandatory?

alisterburt, 2024-10-04:
@pconesa yes the tilt series CTF info is mandatory, we treat it separately to 2D estimation as in the warp pipeline they serve different purposes
- fs CTF is spatially resolved for determining defocus handedness
- ts CTF is a single CTF per tilt and spatial info comes from position in the tomograms, it is determined with extra constraints

pconesa, 2024-10-07:
Thanks @alisterburt , so , I understand that for handedness we need frameseries available somewhere (not averages)?

alisterburt, 2024-10-18:
@pconesa sorry I missed this - you need warp frame series metadata, this can be generated by running `fs_ctf`

closed 2024-10-18 by alisterburt