warpem/warp #265: Any idea why we are getting a "grid of blobs" in the recosntruction of the first TS (TS_01) following the warptools tutorial?

pconesa, 2024-10-03:
[image]

pconesa, 2024-10-03:
To avoid double posting I close this. I've asked the same thing in the google group. Maybe someone else knows.

alisterburt, 2024-10-03:
just tested and runnning fine here on the main branch

```sh
WarpTools ts_reconstruct \
> --settings warp_tiltseries.settings \
> --angpix 10 \
> --perdevice 4 \
> --output_processing recon_test
```
[image]

here is what the xml looks like for TS_1

[TS_1.xml](https://github.com/user-attachments/files/17247937/TS_1.xml.txt)

let's track here - easier than the mailing list

alisterburt, 2024-10-03:
my suspects would be
- something wrong with tilt series alignment data (did `ts_aretomo` or `ts_etomo_*` run ok?)
- something wrong with tilt series ctf data (did you run `ts_ctf`?)

pconesa, 2024-10-04:
Yes, we run both, we also saw that TS_01.xml changed the shifts values from something reasonable after ts_etomo to something suspiciously wrong after ts_ctf.

Can paste the values but from #.some decimals values changes to "huge integers"

I'll ask my colleague to attach the files

pconesa, 2024-10-04:
We will make a quick test... but it may be a "locale" issue...? Machine is in ES (Spanish)... maybe we lost the decimal places due to decimal separator being different.

alisterburt, 2024-10-04:
That would make sense, we've had a couple of locale related bugs - thought we had squashed them all but there might be some left over

pconesa, 2024-10-05:
Can confirm it was a locale (ES) issue.

alisterburt, 2024-10-05:
Thanks for confirming @pconesa - will leave open so we know to look into/fix it ☺️