warpem/warp #266: System.NotImplementedException in MCore/Warp during Refinement

kainvde, 2024-10-03:
Hi,

I’m encountering an issue during the refinement process using M, where I receive the error: System.NotImplementedException: The method or operation is not implemented. This occurs sometimes during the check with --iter 0, and sometimes i get past that and it occurs during the first refinement. 

Interestingly, the process sometimes works when I adjust angpix_resample, but there doesn't seem to be a consistent pattern.

Thanks!
Kain

Examples:

Check:

Loading population... Done
Creating directories... Done
Spawning workers... Done
Preparing for refinement \u2013 this will take a few minutes per species
Preparing refinement requisites...
1/1                                                                                                                                                                                                                                                         
Performing refinement
Preparing population for data source 02...Done
Loading gain reference for 02... Done
Refining all series in data source...
104/104                                                                                                                                                                                                                                                     
Commiting changes in 02...Unhandled exception.Unhandled exception.  System.NotImplementedException: The method or operation is not implemented.
   at MCore.MCore.WorkerDied(Object sender, EventArgs e) in /nfs/scistore16/itgrp/schloegl/src/warp/MCore/MCore.cs:line 597
   at Warp.WorkerWrapper.ReportDeath() in /nfs/scistore16/itgrp/schloegl/src/warp/WarpLib/WorkerWrapper.cs:line 255
   at Warp.WorkerWrapper.<StartHeartbeat>b__14_0() in /nfs/scistore16/itgrp/schloegl/src/warp/WarpLib/WorkerWrapper.cs:line 189
System.NotImplementedException: The method or operation is not implemented.
   at MCore.MCore.WorkerDied(Object sender, EventArgs e) in /nfs/scistore16/itgrp/schloegl/src/warp/MCore/MCore.cs:line 597
   at Warp.WorkerWrapper.ReportDeath() in /nfs/scistore16/itgrp/schloegl/src/warp/WarpLib/WorkerWrapper.cs:line 255
   at Warp.WorkerWrapper.<StartHeartbeat>b__14_0() in /nfs/scistore16/itgrp/schloegl/src/warp/WarpLib/WorkerWrapper.cs:line 189
./M04_check.sh: line 3: 2166237 Aborted                 MCore --population m/02.population --iter 0

First Refinement:

Loading population... Done
Creating directories... Done
Spawning workers... Done
Preparing for refinement \u2013 this will take a few minutes per species
Preparing refinement requisites...
1/1                                                                                                                                                                                                                                                         
Performing refinement
Preparing population for data source 02...Done
Loading gain reference for 02... Done
Refining all series in data source...
104/104                                                                                                                                                                                                                                                     
Commiting changes in 02...Unhandled exception. Unhandled exception. System.NotImplementedException: The method or operation is not implemented.
   at MCore.MCore.WorkerDied(Object sender, EventArgs e) in /nfs/scistore16/itgrp/schloegl/src/warp/MCore/MCore.cs:line 597
   at Warp.WorkerWrapper.ReportDeath() in /nfs/scistore16/itgrp/schloegl/src/warp/WarpLib/WorkerWrapper.cs:line 255
   at Warp.WorkerWrapper.<StartHeartbeat>b__14_0() in /nfs/scistore16/itgrp/schloegl/src/warp/WarpLib/WorkerWrapper.cs:line 189System.NotImplementedException: The method or operation is not implemented.
   at MCore.MCore.WorkerDied(Object sender, EventArgs e) in /nfs/scistore16/itgrp/schloegl/src/warp/MCore/MCore.cs:line 597
   at Warp.WorkerWrapper.ReportDeath() in /nfs/scistore16/itgrp/schloegl/src/warp/WarpLib/WorkerWrapper.cs:line 255
   at Warp.WorkerWrapper.<StartHeartbeat>b__14_0() in /nfs/scistore16/itgrp/schloegl/src/warp/WarpLib/WorkerWrapper.cs:line 189

alisterburt, 2024-10-03:
Hey @kainvde - thanks for the report, this looks like a duplicate of #237 and likely the same issue as #261. I'm going to close those issues in favor of this one and track here

Could you confirm 
- do you see the issue when running with a single GPU
- did you run EstimateWeights at any point

kainvde, 2024-10-04:
Hi Alister,

Thanks for the quick reply. 

I switched to one GPU, and the first refinement worked without the error. However, when running the second refinement on one GPU, I encountered the same error:

./M06_ref3.sh 
Loading population... Done
Creating directories... Done
Spawning workers... Done
Preparing for refinement \u2013 this will take a few minutes per species
Preparing refinement requisites...
1/1                                                                                                                                                
Performing refinement
Preparing population for data source 02...Done
Loading gain reference for 02... Done
Refining all series in data source...
104/104                                                                                                                                            
Commiting changes in 02...Unhandled exception. Unhandled exception. System.NotImplementedException: The method or operation is not implemented.
   at MCore.MCore.WorkerDied(Object sender, EventArgs e) in /nfs/scistore16/itgrp/schloegl/src/warp/MCore/MCore.cs:line 597
   at Warp.WorkerWrapper.ReportDeath() in /nfs/scistore16/itgrp/schloegl/src/warp/WarpLib/WorkerWrapper.cs:line 255
   at Warp.WorkerWrapper.<StartHeartbeat>b__14_0() in /nfs/scistore16/itgrp/schloegl/src/warp/WarpLib/WorkerWrapper.cs:line 189
System.NotImplementedException: The method or operation is not implemented.
   at MCore.MCore.WorkerDied(Object sender, EventArgs e) in /nfs/scistore16/itgrp/schloegl/src/warp/MCore/MCore.cs:line 597
   at Warp.WorkerWrapper.ReportDeath() in /nfs/scistore16/itgrp/schloegl/src/warp/WarpLib/WorkerWrapper.cs:line 255
   at Warp.WorkerWrapper.<StartHeartbeat>b__14_0() in /nfs/scistore16/itgrp/schloegl/src/warp/WarpLib/WorkerWrapper.cs:line 189
./M06_ref3.sh: line 5: 438216 Aborted                 MCore --population m/02.population --refine_imagewarp 6x4 --refine_particles --ctf_defocus

But then switched back to 4 gpus for the second refinement and then it worked.

I haven’t done an EstimateWeights yet.

Thanks,
Kain

alisterburt, 2024-10-04:
Thanks for the useful debugging info, I'll try to reproduce on my end. Could you confirm that the following seems like what you did?

normal WarpTools processing -> refine in WarpTools with >1 GPU -> failure -> refine with 1 GPU -> success -> refine with 1 GPU -> failure -> refine with >1 GPU -> success

kainvde, 2024-10-04:
Yeah that's right. Thanks!

sagarbiophysics, 2024-11-07:
Hi @alisterburt  I had exact issue after running Estimate weights! It looks like you are aware of it. 

Is there a smarter way to recover? I tried to restore hashed version, but the issue seems to persist. Only way I managed to do is re-runing Warptools project.

sagarbiophysics, 2024-12-04:
@alisterburt update: So recovery was easy after just rerunning and cleaning data sources. 
However, as I am running quite a few calculations for b-factor analysis. I am repeatedly running into System.NotImplementedException errors at random. It run's through most times, but crashes on some populations. Rerunning on crashes on some, succeeds on others. I am using same configuration 4 x NVIDIA A40 GPUs. Also tried 1X, which as similar erratic behaviour! I guess it's indeed something with memory, but i am a bit clueless with rather unpredictable and erratic fails, but then again that's to expect with memory evils!! 
Hope you are doing great! Will also push out TOMOMAN module for warptools soon :)

jmdobbs, 2025-01-16:
Hi @alisterburt and all, we are increasingly getting the same thing very often during refinement. It seems to be independent of the number of GPUs for us. It may have something to do with GPU VRAM (which slurm can't specify) being used by other users of the same GPU node, but I'm not certain. It seems almost random as to when this occurs, can happen even when multiple M runs are submitted in the same script but the preceding one and/or subsequent one run through, and can even happen with --iter 0. Possibly it just happens when someone else runs something demanding on the same node? It seems to happen less when our cluster is not busy.

Similarly, we have noticed failures when M is run by two users on the same node, so there is some crossover happening somehow that may be leading to this.

alisterburt, 2025-01-16:
@jmdobbs interesting, thank you - we definitely haven't seen this and always run with entire GPUs reserved rather than part of a GPU... that sounds messy and I wouldn't be surprised if it causes issues

The crosstalk suggestion is interesting, this is when multiple WarpWorker/MCore processes are running on the same GPU?

alisterburt, 2025-01-16:
@jmdobbs gotcha this makes sense, if it is M specific I bet it's something to do with the way pinned memory is used... I'll ask Dimitry about this today and get back to you

mstabrin, 2026-02-09:
Hello @alisterburt,

Hope you are doing great :) 
we are also now running into this exact issue.
Did you get any information back in the days?

Best,
Markus