warpem/warp #267: Questions for integrtion in Scipion

pconesa, 2024-10-07:
We are trying to integrate several warp methods in scipion. Ideally we want "atomized" methods. This is to allow integration at any steps but we are struggling to to so for some methods and maybe is not possible?.

For example, we have successfully integrated: ts_ctf by generating all required data and metadata for this process

we are generating "averages" folder unsatcking our tilt series stacks, generating tomostar and calling create-settings

Now, we wanted to integrate others this way for example ts_reconstruct....and here we have a doubt. In Scipion we have Tiltseries (data and metadata) and CTF(metadata).... Our ctfs are very simple, deficusU, V per tilt..... but We have seen that the CTF metadata used by ts-recostruct  may be more complex that just defocusU and DefocusV per title image or a something inferred from these values?

The TS_##.xml has several sections attributes we are now sure how critical they are:

AreAnglesInverted="False" PlaneNormal="0.0713941, -0.08116225, 0.9941406" Bfactor="0" Weight="1" MagnificationCorrection="1, 0, 0, 1" UnselectFilter="False" UnselectManual="" CTFResolutionEstimate="4.5"

And there are other attributes like:

TiltPS1D, PS1D, GridMovementX, GridCTF , ....

We are guessing that with defocusU and V (as if they were generated by ctffind4) we can populate GridCTF, GridCTFDefocusAngle, GridCTFDefocusDelta?

But what about the rest? are they key to the reconstruct process?

We need you advice here. Another option we are thinking is to merge the ts_ctf and ts_reconstruct in one single step.

We also tried to integrate ts_defocus_hand--- but seems to need from fs_ctf? And fs_ctf from frameseries...?

If so we need to provide frameseries too which is not ideal. So my question here would be if there is a way to run the fs_ctf on averages or it needs frameseries?

alisterburt, 2024-10-07:
Hi @pconesa !

I understand your desire for 'atomized' processing and wish it were that way already - thank you for your patience when wiring things together

> We are guessing that with defocusU and V (as if they were generated by ctffind4) we can populate GridCTF, GridCTFDefocusAngle, GridCTFDefocusDelta?

Yes that's correct

> But what about the rest? are they key to the reconstruct process?

I took a look at the reconstruction method and don't see it using anything else, you should be good to go 🙂 - let me know if you run into issues

> We also tried to integrate ts_defocus_hand--- but seems to need from fs_ctf? And fs_ctf from frameseries...?
>
> If so we need to provide frameseries too which is not ideal. So my question here would be if there is a way to run the fs_ctf on averages or it needs frameseries?

You can run `fs_ctf` on averages, you should run it with a 2x2x1 grid to get the necessary information for defocus handedness checking

pconesa, 2024-10-07:
Nice, thanks Alister for your quick reply...

It is no clear to me how to run the fs_ctf on averages 8 think we tried
usesum Param with no success.

Is there any example we can follow?

pconesa, 2024-10-08:
We managed to call fs_ctf on averages and then ts_defocus_hand. 

We are currently doing ts_ctf  and ts_reconstruct merged in one step but will try separating them. So far so good.

closed 2024-10-08 by pconesa