warpem/warp #268: ts_import_alignments not working for .aln

DrJesseHansen, 2024-10-07:
hi,

I am running ver dev 28 warptools.  

Is it expected behaviour that ts_import_alignments requires .xf files and cannot read in .aln instead?   I created a folder for the tilt series and dropped in the .aln file.  I ran the command below:

```
WarpTools ts_import_alignments \
--settings warp_tiltseries.settings \
--alignments ALN_folder \
--alignment_angpix 13.584
```

I got the error below:

`Parsing previous results for each item, if available...
1/1, previous metadata found for 1                                                                                             
Failed to import alignments for TS_87.tomostar: Could not find TS_87.xf
`

I then instead dropped the .xf file and .tlt file into that folder and the job ran through.  

Is WarpTools not able to read in .aln files?

J

shahpnmlab, 2024-10-07:
WarpTools does not support .aln files. You will have to get AreTomo2 to somehow spit out the xf files for you.. but this is currently not possible due to a bug in AT2 : https://github.com/czimaginginstitute/AreTomo2/issues/28

DrJesseHansen, 2024-10-07:
thank you Pranav,  

I do have the xf files actually, I guess from an earlier version before the bug arose?   However I can't use those xf files directly in warptools since the scaling is different since I ran Aretomo on bin1 stacks from Desktop warp, while Linux Warp uses bin 8 stacks by default.  [Posted this to google group](https://groups.google.com/g/warp-em/c/yX_CJOLqpio)

alisterburt, 2024-10-07:
@DrJesseHansen as pranav said there is no support for aln files - `ts_import_alignments` has an `--alignment_angpix` option https://warpem.github.io/warp/reference/warptools/api/tilt_series/#ts_import_alignments for dealing with this

closed 2024-10-07 by alisterburt