warpem/warp #269: Importing alignments into Warp

jowagner91, 2024-10-08:
Hello,

I am currently trying to use Warp to reprocess data for an ongoing project that involves a lot of manual segmentation as well. All the segmentation work was done on tomograms corrected for X-tilt in Etomo.

When I import the .xf and .tlt files into Warp, all the reconstructions are done without X-tilt correction. Is there a way to reconstruct tomograms in Warp with X-tilt correction? This would ensure the coordinates match those from segmentation. Otherwise, I would need to figure out the transformations manually after the fact.

alisterburt, 2024-10-08:
Hi @jowagner91 - X tilt is unfortunately not part of the Warp model

jowagner91, 2024-10-09:
Thanks for the quick reply. There’s no way to apply shifts in the z-position of the tomogram to recenter it, is there?

mathewmclaren, 2024-10-09:
Hi @jowagner91, if you did your segmentation in IMOD then you can use the command imodtrans to translate, scale and rotate your model to match the new tomogram reconstruction.

alisterburt, 2024-10-09:
> Thanks for the quick reply. There’s no way to apply shifts in the z-position of the tomogram to recenter it, is there? 

@jowagner91 z-shift is a change in the shifts that center each image so it's taken into account if it's in the XF... unsure if IMOD puts it in the xf or applies as an extra transform during reconstruction though 🤷‍♂️

Thanks for the input @mathewmclaren !

alisterburt, 2024-10-09:
Going to close as we have no plans to add this

closed 2024-10-09 by alisterburt