warpem/warp #270: Box size limit due to csharp array index limit

deadlift385, 2024-10-09:
Hi Alister,

I'm pushing the box size on an A100 with 80G GPU and I noticed that there is limit because of a csharp array indexing limit when running MTools create_species.

In Helper.cs, line 1104, in the "public static byte[] ToBytes(float[] data)" function we have:

byte[] Bytes = new byte[data.Length * sizeof(float)];

if "data.Length * sizeof(float)" is larger than Int32.MaxValue or 2147483647 it crashes, because of the csharp array index limit.

Basically:
byte[] Bytes = new byte[Int32.MaxValue];
--> this works
byte[] Bytes = new byte[Int32.MaxValue + 1];
--> this fails

So if the box is larger than about 812 pixels, a crash results:
>>> np.power(2147483647/4.,1/3.) = 812.7

ToBytes(float[] data) is called in Species.cs:

Builder.Append(MathHelper.GetSHA1(Helper.ToBytes(HalfMap1.GetHostContinuousCopy())));
Builder.Append(MathHelper.GetSHA1(Helper.ToBytes(HalfMap2.GetHostContinuousCopy())));
Builder.Append(MathHelper.GetSHA1(Helper.ToBytes(Mask.GetHostContinuousCopy())));

where the whole half maps and the mask are dumped into a linear array just for calculating a hash value.

I suggest to change the way the hash is calculated to avoid the array index limit. E.g. for a quick fix I changed to code to:

        public static byte[] ToBytes_For_Hash(float[] data)
        {
            byte[] Bytes = new byte[data.Length / 100 * sizeof(float)];
            Buffer.BlockCopy(data, 0, Bytes, 0, Bytes.Length / 100);
            return Bytes;
        }

and then

Builder.Append(MathHelper.GetSHA1(Helper.ToBytes_For_Hash(HalfMap1.GetHostContinuousCopy())));
Builder.Append(MathHelper.GetSHA1(Helper.ToBytes_For_Hash(HalfMap2.GetHostContinuousCopy())));
Builder.Append(MathHelper.GetSHA1(Helper.ToBytes_For_Hash(Mask.GetHostContinuousCopy())));

which seems to work.

Thanks and all the best, Simon

alisterburt, 2024-10-09:
@deadlift385 thanks for the report and the code snippet, super useful! Although FYI I think you're copying less data than you should be, shouldn't it be...

```csharp
public static byte[] ToBytes_For_Hash(float[] data)
{
    byte[] Bytes = new byte[data.Length / 100 * sizeof(float)];
    Buffer.BlockCopy(data, 0, Bytes, 0, Bytes.Length / 100);
    return Bytes;
}
```

I'll check with Dimitry before pushing a fix as he's mentioned only hashing a small portion of the data to me before and maybe we can/should reuse the same mechanism...

deadlift385, 2024-10-09:
Yes, can you just hash the central section of the map?

True, I guess it should be:

public static byte[] ToBytes_For_Hash(float[] data)
{
    byte[] Bytes = new byte[data.Length / 100 * sizeof(float)];
    Buffer.BlockCopy(data, 0, Bytes, 0, Bytes.Length);
    return Bytes;
}

alisterburt, 2024-10-09:
Data hashes are calculated like this for `Movie` and `TiltSeries`
https://github.com/warpem/warp/blob/8939b18a9d2025cbf4aa71a61107b1623bfcde7a/WarpLib/Movie.cs#L1167-L1179
https://github.com/warpem/warp/blob/8939b18a9d2025cbf4aa71a61107b1623bfcde7a/WarpLib/TiltSeries.cs#L11251-L11265

I've gone with adding
```csharp
public static byte[] ToBytesSafe(float[] data)
{
    // truncates data in byte array if necessary
    byte[] Bytes = new byte[Math.Min(1 << 19, data.Length * sizeof(float))];
    Buffer.BlockCopy(data, 0, Bytes, 0, Bytes.Length);

    return Bytes;
}
```

alisterburt, 2024-10-09:
fix deploying in `v2.0.0dev29` https://github.com/warpem/warp/actions/runs/11260129066

closed 2024-10-09 by alisterburt

deadlift385, 2024-10-09:
Great! Thank you very much. The maps we are getting are absolutely stunning!

alisterburt, 2024-10-09:
@deadlift385 would love to hear about what you're working on - feel free to drop me an email if you're comfortable sharing :-)

PR #271 (2024-10-09): Use safe version of Helper.ToBytes when calculating data hashes in Species.cs
closes #270 

this should maintain identical behavior to what was there before and also work for large boxes

cc @deadlift385