warpem/warp #276: M reconstruction deteriorates after shifting species

deadlift385, 2024-10-17:
Hi Alister,

I cannot get the same quality reconstruction after shifting a species.

Step 1) Start with an M refined reconstruction at 3.6 A resolution (box = 750, apix = 1.6).

Step 2a) Shift the species (MTools shift_species), which gives me half maps that look fine and are centered as intended.
Step 2b) Calculate the FSC after cropping out a box (256) at the center of the shifted half maps and applying a spherical mask. The resolution is 3.9 A, as expected.

Step 3a) Run MCore without refinement.
Step 3a) Calculate the FSC after cropping out a box (256) at the center of the shifted half maps and applying a spherical mask. The resolution is 6.7 A, much worse.

# Step 1)
MCore \
--population m_run01/population_01.population \
--iter 0
#2024-10-17 18:32:16.890 Global resolution is 3.597

# Step 2a)
MTools shift_species \
--population m_run01/population_01.population \
--species m_run01/species/species_01_61d3e558/species_01.species \
--x 132 --y 298 --z 194
# Step 2b)
# Resolution from half maps after cropping and applying spherical mask = 3.9 (calculated with eman2 and relion)

# Step 3a)
MCore \
--population m_run01/population_01.population \
--iter 0
# Step 3b)
# Resolution from half maps after cropping and applying spherical mask = 6.7 (calculated with eman2 and relion)

It's not obvious to me why one would get a substantially worse reconstruction (locally) after shifting the species and without refining anything. Am I missing something here? Is there some hidden weighing going on in the reconstruction step that gets messed up after shifting the species?

All the best, Simon

alisterburt, 2024-10-18:
Hi @deadlift385 ! Thanks for the report - I'm not familiar with the move_species code myself so will have to take a look but I agree with your assessment, I don't see why it would manage to make a reconstruction after shifting the species then fail when running MCore...

In the meantime, you could try applying shifts to your input STAR file for species creation following the logic here

https://gist.github.com/alisterburt/5fa3e867f8004171fff57579b379ffa9

This script allows for applying multiple transformations at once, yielding n output particles per input particle. If you want to skip the re-orientation supply an identity matrix or comment out the  code 🙂

deadlift385, 2024-10-18:
Thank you for the link to your script! I can make Relion star files after using MTools expand_symmetry and MTools shift_species, this works all fine. After symmetry expansion M gives me the same reconstruction as compared to imposing symmetry (expected). Getting the same quality reconstructions, and exported subtomograms for classification, at shifted positions is the issue. I can do a new species creation, but I would have to see how stable the M refinement of those would be.

alisterburt, 2024-10-18:
@deadlift385 right, I'm suggesting that you remove the use of M integrated tools, apply the shifts yourself then create a new species in M and see if the problems persist - this would allow us to say whether the problem is with `shift_species` or something strange with your data (unlikely) 🙂

deadlift385, 2024-10-18:
In fact, I have already done this, created a new species with the new positions and angles, then run M without refinement (--iter 0). Same resolution as after shifting the original species, 6.4 instead of 3.9. And the maps look correspondingly worse.

alisterburt, 2024-10-18:
@deadlift385 hmm, could you share a few xy slices of the maps? It seems like the resolution must be worse where your new map is centered... the only missing control would be to check the FSC of your newly centered region in the original map

deadlift385, 2024-11-15:
I ran a few more tests regarding this issue.

My conclusion is that if you shift away (a substantial distance of approximately 300 A here) from the particle positions where the distortion model was originally refined, the distortion model at the new shifted positions is just not very accurate any more (there are not many particles per tomogram here), thus if one reconstructs with M, the resolution gets substantially worse. If I repeat the entire M refinement after shifting (e.g. with a mask and updated mask and the center of mass of the particles way off the center of the box and part of the particles outside of the box), I pretty much restore the resolution.

So make a new round of refinement if the shift is large and if there are only few particles to refine the distortion model.

I guess this issue could be closed. Thanks again, Simon

alisterburt, 2024-11-15:
@deadlift385 thanks for circling back, this is useful info but feels somewhat surprising - what spatial resolution were you using to fit image warp?

closed 2024-11-15 by alisterburt

deadlift385, 2024-11-15:
It was 4x4. The last refinement before shifting was:

MCore \
--population m_run01/population_01.population \
--refine_imagewarp 4x4 \
--refine_particles \
--refine_stageangles \
--refine_volumewarp 4x4x2x10 \
--ctf_defocus \
--refine_tiltmovies \
--refine_mag \
--ctf_cs \
--perdevice_refine 1

with 3 sampled trajectories.

deadlift385, 2024-11-15:
E) FSC in mask of full particle (what is within the box)
F) FSC in smaller spherical mask at center of box
after just shifting the input map, after reconstruction with M, after refinement and reconstruction with M

[image]

alisterburt, 2024-11-15:
> It was 4x4. The last refinement before shifting was:
>
> MCore --population m_run01/population_01.population --refine_imagewarp 4x4 --refine_particles --refine_stageangles --refine_volumewarp 4x4x2x10 [...]

interesting, 4x4 isn't too crazy but there is potential for overfitting. If you by 3 sampled trajectories you mean 3 points of temporal resolution for particle poses then this is a very likely source of overfitting - I'm glad you were able to fix it with another round of M. If I were processing this myself I would go back and start from more conservative M refinements in case many particles ended up outside the radius of convergence for the optimisation

deadlift385, 2024-11-18:
I don't think there is any overfitting. The giga Da particles are so large that aligning noise is almost impossible. Going from 1 to 3 points of temporal resolution for particle poses does pretty much nothing here. Also the maps look as they should given the nominal resolution.

With 1 point:
#2024-10-28 20:04:09.873 Global resolution is 3.323
With 3 points:
#2024-11-01 16:58:32.421 Global resolution is 3.324

deadlift385, 2024-11-18:
I think the poses at each control point are identical, given the standard deviations for each particle:

[image]

I can check the CTF values. Also I could calculate the FSC with a "perfect" model (2.4 A structure from 2D data) and see if the nominal resolution at 0.5 corresponds to the value determined from the half maps.