warpem/warp #277: ts_export_particles cannot create star file

hamid13r, 2024-10-17:
Hi,

 I am trying to extract some sunvolumes in WarpTools and I am seeing a weird error.   The extraction runs, all the subvolumes are created but I get this error:
```
No particles to write out. Check that rlnMicrographName entries match tomostar file for each tilt series.  
```
Full error message:
[ts_export_output.txt](https://github.com/user-attachments/files/17423909/ts_export_output.txt)

And there is no starfile created at the end. This is weird because if they cannot be found I don't understand how did they get created? They are all created correctly since I can run relion with the same dataset when I use the starfile from Warp1. 

Is it possible that my ts name "JTL016A_2_L01_ts_002.mrc.tomostar" is the problem because it has .mrc in it? This is the only difference I see between my dataset and the tutorial.

alisterburt, 2024-10-17:
Hey @hamid13r - can you run with `WARP_DEBUG=1` and send over what it prints out?

hamid13r, 2024-10-17:
I ran this command:
`export WARP_DEBUG=1; WarpTools ts_export_particles --settings warp_tiltseries.settings  --input_star test.star --coords_angpix 13.28 --output_star test.star --output_angpix 13.28 --box 36 --diameter 300 --relative_output_paths --3d
`

and the output file and logs are attached. A little unsure if I did it right because the outputs look the same!
[tomostar_log.txt](https://github.com/user-attachments/files/17424301/tomostar_log.txt)
[worker0_out.txt](https://github.com/user-attachments/files/17424302/worker0_out.txt)
[warp_ts_export_debug.txt](https://github.com/user-attachments/files/17424303/warp_ts_export_debug.txt)

alisterburt, 2024-10-18:
Hi @hamid13r - the debug output is quite clear... warptools is not finding particles and you need to check your input

```
0/52Processing JTL016A_2_L11_ts_005.mrc.tomostar
Processing JTL016A_2_L21_ts_003.mrc.tomostar
Processing JTL016A_2_L10_ts_004.mrc.tomostar
no particles found in JTL016A_2_L10_ts_004.mrc.tomostar, skipping...
no particles found in JTL016A_2_L21_ts_003.mrc.tomostar, skipping...
no particles found in JTL016A_2_L11_ts_005.mrc.tomostar, skipping...
3/52, 00:01 remainingProcessing JTL016A_2_L02_ts_005.mrc.tomostar
no particles found in JTL016A_2_L02_ts_005.mrc.tomostar, skipping...
4/52, 00:01 remainingProcessing JTL016A_2_L15_ts_002.mrc.tomostar
no particles found in JTL016A_2_L15_ts_002.mrc.tomostar, skipping...
Processing JTL016A_2_L21_ts_002.mrc.tomostar
no particles found in JTL016A_2_L21_ts_002.mrc.tomostar, skipping...
5/52, 00:00 remainingProcessing JTL016A_2_L03_ts_006.mrc.tomostar
no particles found in JTL016A_2_L03_ts_006.mrc.tomostar, skipping...
7/52, 00:00 remainingProcessing JTL016A_2_L19_ts_005.mrc.tomostar
no particles found in JTL016A_2_L19_ts_005.mrc.tomostar, skipping...
8/52, 00:00 remainingProcessing JTL016A_2_L17_ts_003.mrc.tomostar
no particles found in JTL016A_2_L17_ts_003.mrc.tomostar, skipping...
9/52, 00:00 remainingProcessing JTL016A_2_L06_ts_004.mrc.tomostar
no particles found in JTL016A_2_L06_ts_004.mrc.tomostar, skipping...
10/52, 00:00 remainingProcessing JTL016A_2_L17_ts_006.mrc.tomostar
no particles found in JTL016A_2_L17_ts_006.mrc.tomostar, skipping...
11/52, 00:00 remainingProcessing JTL016A_2_L17_ts_002.mrc.tomostar
no particles found in JTL016A_2_L17_ts_002.mrc.tomostar, skipping...
12/52, 00:00 remainingProcessing JTL016A_2_L03_ts_004.mrc.tomostar
no particles found in JTL016A_2_L03_ts_004.mrc.tomostar, skipping...
13/52, 00:00 remainingProcessing JTL016A_2_L21_ts_004.mrc.tomostar
no particles found in JTL016A_2_L21_ts_004.mrc.tomostar, skipping...
14/52, 00:00 remainingProcessing JTL016A_2_L19_ts_002.mrc.tomostar
no particles found in JTL016A_2_L19_ts_002.mrc.tomostar, skipping...
15/52, 00:00 remainingProcessing JTL016A_2_L05_ts_004.mrc.tomostar
no particles found in JTL016A_2_L05_ts_004.mrc.tomostar, skipping...
16/52, 00:00 remainingProcessing JTL016A_2_L02_ts_004.mrc.tomostar
no particles found in JTL016A_2_L02_ts_004.mrc.tomostar, skipping...
17/52, 00:00 remainingProcessing JTL016A_2_L06_ts_003.mrc.tomostar
no particles found in JTL016A_2_L06_ts_003.mrc.tomostar, skipping...
Processing JTL016A_2_L01_ts_007.mrc.tomostar
no particles found in JTL016A_2_L01_ts_007.mrc.tomostar, skipping...
19/52, 00:00 remainingProcessing JTL016A_2_L15_ts_003.mrc.tomostar
no particles found in JTL016A_2_L15_ts_003.mrc.tomostar, skipping...
20/52, 00:00 remainingProcessing JTL016A_2_L19_ts_003.mrc.tomostar
no particles found in JTL016A_2_L19_ts_003.mrc.tomostar, skipping...
21/52, 00:00 remainingProcessing JTL016A_2_L21_ts_005.mrc.tomostar
no particles found in JTL016A_2_L21_ts_005.mrc.tomostar, skipping...
22/52, 00:00 remainingProcessing JTL016A_2_L02_ts_002.mrc.tomostar
no particles found in JTL016A_2_L02_ts_002.mrc.tomostar, skipping...
23/52, 00:00 remainingProcessing JTL016A_2_L11_ts_003.mrc.tomostar
no particles found in JTL016A_2_L11_ts_003.mrc.tomostar, skipping...
24/52, 00:00 remainingProcessing JTL016A_2_L01_ts_008.mrc.tomostar
no particles found in JTL016A_2_L01_ts_008.mrc.tomostar, skipping...
25/52, 00:00 remainingProcessing JTL016A_2_L19_ts_006.mrc.tomostar
no particles found in JTL016A_2_L19_ts_006.mrc.tomostar, skipping...
Processing JTL016A_2_L01_ts_004.mrc.tomostar
no particles found in JTL016A_2_L01_ts_004.mrc.tomostar, skipping...
27/52, 00:00 remainingProcessing JTL016A_2_L11_ts_006.mrc.tomostar
no particles found in JTL016A_2_L11_ts_006.mrc.tomostar, skipping...
28/52, 00:00 remainingProcessing JTL016A_2_L03_ts_005.mrc.tomostar
no particles found in JTL016A_2_L03_ts_005.mrc.tomostar, skipping...
Processing JTL016A_2_L01_ts_002.tomostar
no particles found in JTL016A_2_L01_ts_002.tomostar, skipping...
30/52, 00:00 remainingProcessing JTL016A_2_L01_ts_006.mrc.tomostar
no particles found in JTL016A_2_L01_ts_006.mrc.tomostar, skipping...
31/52, 00:00 remainingProcessing JTL016A_2_L19_ts_004.mrc.tomostar
no particles found in JTL016A_2_L19_ts_004.mrc.tomostar, skipping...
32/52, 00:00 remainingProcessing JTL016A_2_L15_ts_005.mrc.tomostar
no particles found in JTL016A_2_L15_ts_005.mrc.tomostar, skipping...
33/52, 00:00 remainingProcessing JTL016A_2_L05_ts_003.mrc.tomostar
[... 12 lines cut ...]
no particles found in JTL016A_2_L01_ts_003.mrc.tomostar, skipping...
40/52, 00:00 remainingProcessing JTL016A_2_L10_ts_002.mrc.tomostar
no particles found in JTL016A_2_L10_ts_002.mrc.tomostar, skipping...
41/52, 00:00 remainingProcessing JTL016A_2_L17_ts_005.mrc.tomostar
no particles found in JTL016A_2_L17_ts_005.mrc.tomostar, skipping...
42/52, 00:00 remainingProcessing JTL016A_2_L05_ts_002.mrc.tomostar
no particles found in JTL016A_2_L05_ts_002.mrc.tomostar, skipping...
Processing JTL016A_2_L03_ts_003.mrc.tomostar
no particles found in JTL016A_2_L03_ts_003.mrc.tomostar, skipping...
44/52, 00:00 remainingProcessing JTL016A_2_L03_ts_002.mrc.tomostar
no particles found in JTL016A_2_L03_ts_002.mrc.tomostar, skipping...
Processing JTL016A_2_L01_ts_005.mrc.tomostar
no particles found in JTL016A_2_L01_ts_005.mrc.tomostar, skipping...
46/52, 00:00 remainingProcessing JTL016A_2_L17_ts_004.mrc.tomostar
[... cut ...]
no particles found in JTL016A_2_L06_ts_002.mrc.tomostar, skipping...
48/52, 00:00 remainingProcessing JTL016A_2_L22_ts_003.mrc.tomostar
no particles found in JTL016A_2_L22_ts_003.mrc.tomostar, skipping...
49/52, 00:00 remainingProcessing JTL016A_2_L17_ts_007.mrc.tomostar
no particles found in JTL016A_2_L17_ts_007.mrc.tomostar, skipping...
50/52, 00:00 remainingProcessing JTL016A_2_L01_ts_002.mrc.tomostar
185 particles for JTL016A_2_L01_ts_002.mrc.tomostar
Processing JTL016A_2_L11_ts_004.mrc.tomostar
```

hamid13r, 2024-10-19:
Oh I am sorry I should have clarified that, this was a test with a smaller star file and the same happens with the full star file. Nevertheless, it is often the case that we don't have particles from all the tomograms. there are 185 particles in `JTL016A_2_L01_ts_002.mrc.tomostar` that I am expecting to get an output star file for.

alisterburt, 2024-10-21:
honestly @hamid13r this report doesn't feel clear to me here
- what is the contents of `rlnMicrographName` in the input STAR file
- what is the name of the corresponding tomostar file
- what is the behavior that is unexpected

hamid13r, 2024-10-24:
Sorry. Let me try to clarify the whole issue. (going by numbers to be able to reference) 
1.
The dataset is already processed in Warp1.1 and I just made the settings files and it recognizes my tomograms. And new xml files are created:

`ls warp_tiltseries/
JTL016A_2_L01_ts_002.mrc.xml
JTL016A_2_L01_ts_003.mrc.xml
JTL016A_2_L01_ts_004.mrc.xml
....
`
2.
Here is what the input star file looks like:
`
data_

loop_
_rlnCoordinateX #1
_rlnCoordinateY #2
_rlnCoordinateZ #3
_rlnAngleRot #4
_rlnAngleTilt #5
_rlnAnglePsi #6
_rlnLCCmax #7
_rlnCutOff #8
_rlnSearchStd #9
_rlnDetectorPixelSize #10
_rlnMicrographName #11
945     1126    202     110.916092      159.102427      -51.674991      0.434001        0.010000        0.048260        6.650000        JTL016A_2_L01_ts_002.mrc.tomostar
868     1352    398     61.978095       166.224224      -118.071727     0.418502        0.010000        0.048260        6.650000        JTL016A_2_L01_ts_002.mrc.tomostar
[and the rest of the particles]
`
3.
When I run 
`WarpTools ts_export_particles --settings warp_tiltseries.settings  --input_star test.star --coords_angpix 13.28 --output_star test.star --output_angpix 13.28 --box 36 --diameter 300 --relative_output_paths --3d --WARP_DEBUG 1`
I get all the particles created in 
`warp_tiltseries/subtomo/`
but I get the error I copied in the previous post and I don't get an output star file with paths of images (_rlnImageName) and other info. 

4.
I know the particles are created properly, because when I use the output star file that is created in Warp1.1 with the same input I get the same reconstruction. 

So I get the particles (sutomos) but not a starfile that goes with them.

alisterburt, 2024-10-24:
I can't reproduce the problem you're seeing... 3d particle export works well for me and produces an output STAR file

I'm following now, copying relevant parts of logs for clarity - in future please try to trim things down to something minimal so it's easier to work with you on fixing issues

error in master process:
```
Constructing output table for JTL016A_2_L01_ts_002.mrc.tomostar
Failed to process /scratch2/users/hamid/Krios_Data/JTL016A_2_TEM_230503/warp_tiltseries/JTL016A_2_L01_ts_002.mrc.tomostar, marked as unselected
Check logs in /scratch2/users/hamid/Krios_Data/JTL016A_2_TEM_230503/warp_tiltseries/logs for more info.
```

worker log:
```
2024-10-17 23:12:31.901 Received "TomoExportParticleSubtomos", with 4 arguments, for GPU #2, 9996 MB free:
2024-10-17 23:13:01.358 Exported 185 particles for /scratch2/users/hamid/Krios_Data/JTL016A_2_TEM_230503/warp_tiltseries/JTL016A_2_L01_ts_002.mrc.tomostar
2024-10-17 23:13:01.358 Execution took 29.452 seconds
2024-10-17 23:13:01.359
```

worker log contains no record of error/exception...

alisterburt, 2024-10-24:
can you please repeat your test but with a `--output_processing export_test` so that we can confirm the worker log you see is actually from this run?

hamid13r, 2024-10-25:
Thank you, the new run creates this file:

`
cat /scratch2/users/hamid/Krios_Data/JTL016A_2_TEM_230503/export_test/logs/JTL016A_2_L01_ts_002.mrc.log
`
and this is what is inside it:
`
2024-10-25 22:15:51.794 Received "TomoExportParticleSubtomos", with 4 arguments, for GPU #0, 24029 MB free:
2024-10-25 22:15:55.332 Exported 185 particles for /scratch2/users/hamid/Krios_Data/JTL016A_2_TEM_230503/export_test/JTL016A_2_L01_ts_002.mrc.tomostar
2024-10-25 22:15:55.332 Execution took 3.534 seconds
2024-10-25 22:15:55.333
`

hamid13r, 2025-08-27:
> Hi! I have a very similar problem I think, but I couldn't really fix it..
>
> I ran this command:
>
> WarpTools ts_export_particles
> --settings ts.settings
> --input_directory tiltseries/matching/
> --input_pattern *Position_2_3.star
> --input_data tomostar/Position_2_3.tomostar
> --coords_angpix 2.33
> --output_star matching.star
> --output_angpix 2.33
> --box 48
> --diameter 110
> --relative_output_paths
> --2d \
>
> My setting file (ts.settings) looks like this:
>
> ...
>
> My star file (tiltseries/matching/Position_2_3.star) looks like this:
>
> data_
>
> [... 68 lines cut ...]
> 84.700 92.020004 23.828 0.000
>
> The star file wasn't directly from warp picking but from GAPSTOP, but I
> think I really made all the formating as the warp would require. Could you
> help me please? Thanks a lot!!!

My problem turned out to be having a very messy directory with old and new
files. When I made a new one and cleaned up it started working! Hope that
helps!

jychoi0616, 2025-08-28:
Oh! It's 27944 particles and only one tilt series for this job. Are these many?
Would this become a doable job if I do this job in bin4 (pixel size of 9.32) rather than bin1 (pixel size of 2.33) like currently?

jychoi0616, 2025-08-28:
Okay, so what kind of particle numbers would be reasonable to break down to in this case?

jychoi0616, 2025-08-28:
For others who might have the similar issue as mine, after trying a few different numbers of particles, it seems 22k particles are okay in our system!

jychoi0616, 2025-08-28:
It worked well with --2d option.
However, with --3d option actually, even with 130 particles it couldn't finish the job. and the output of WARP_DEBUG was:

------
Running on GPU #0 (39900 MB free), port 42841

Loaded gain reference and defect map: 4096, 4096, 1, normal, real, 1 A/px, ID = -1, False, True, False, 
Execution took 1.074 seconds

Received "TomoExportParticleSubtomos", with 4 arguments, for GPU #0, 39900 MB free:
Exported 130 particles for /ceph/users/bcw78687/dorothy/20250412/J271-3/TS_02/tiltseries/Position_2_3.tomostar
Execution took 18.678 seconds

10 seconds without heartbeat, exiting
--------

I'm not sure what went wrong. The surface output error message was this:
------------
130 particles for Position_2_3.tomostar
Sending export options to worker for Position_2_3.tomostar
Constructing output table for Position_2_3.tomostar
Failed to process /ceph/users/bcw78687/dorothy/20250412/J271-3/TS_02/tiltseries/Position_2_3.tomostar, marked as unselected                                                                        
Check logs in /ceph/users/bcw78687/dorothy/20250412/J271-3/TS_02/tiltseries/logs for more info.
Use the change_selection WarpTool to reactivate this item if required.
1/1, 1 failed, 00:00 remaining                                                                                                                                                                     
Finished processing in 00:00:19
Unhandled exception. System.InvalidOperationException: No particles to write out. Check that rlnMicrographName entries match tomostar file for each tilt series.
   at WarpTools.Commands.ExportParticlesTiltseries.WriteOutputStarFile(Dictionary`2 perTiltSeriesTables, String particleStarPath, Int32 outputDimensionality, Int32 maxMissingTilts) in /usr/share/miniconda/envs/package-build/conda-bld/warp_1737588682413/work/WarpTools/Commands/Tiltseries/ExportParticlesTiltseries.cs:line 1048
   at WarpTools.Commands.ExportParticlesTiltseries.Run(Object options) in /usr/share/miniconda/envs/package-build/conda-bld/warp_1737588682413/work/WarpTools/Commands/Tiltseries/ExportParticlesTiltseries.cs:line 383
   at WarpTools.WarpTools.Run(Object options) in /usr/share/miniconda/envs/package-build/conda-bld/warp_1737588682413/work/WarpTools/Program.cs:line 30
   at Warp.Tools.CommandLineParserHelper.ParseAndRun(String[] args, Func`2 run, Type[] verbs, String appName) in /usr/share/miniconda/envs/package-build/conda-bld/warp_1737588682413/work/WarpLib/Tools/CommandLineParserHelper.cs:line 26
   at WarpTools.WarpTools.Main(String[] args) in /usr/share/miniconda/envs/package-build/conda-bld/warp_1737588682413/work/WarpTools/Program.cs:line 17
   at WarpTools.WarpTools.<Main>(String[] args)
8_ts_export_particles_test: line 14: 1539832 Aborted                 (core dumped) WarpTools ts_export_particles --settings ts.settings --input_directory tiltseries/matching/ --input_pattern *Position_2_3_test.star --input_data tomostar/Position_2_3.tomostar --coords_angpix 2.33 --output_star matching_3d.star --output_angpix 2.33 --box 48 --diameter 110 --relative_output_paths --3d
--------------------