warpem/warp #278: Can't fetch latest version from conda channel?

rdrighetto, 2024-10-18:
Hi,

While helping @phaips with #28, I tried to update the conda `warp` package from `2.0.0dev25` to `2.0.0dev29`.
I tried using `conda update warp` in our existing environment as well as installing a complete new environment from scratch following the [instructions](https://github.com/warpem/warp#linux). I always end up stuck with `2.0.0dev25`.

Is `2.0.0dev29` even available from the `warpem` channel, or building it myself is the only way to go?

Thank you!

alisterburt, 2024-10-18:
https://anaconda.org/warpem/warp/files

dev29 showing as available here, not immediately sure what has changed since 25...

rdrighetto, 2024-10-21:
Hi @alisterburt! I considered that too, but would be weird that I can still see the channel, just not all versions of the package 🤔 

I managed to circumvent the issue by installing `2.0.0dev29` as follows:

`mamba install https://anaconda.org/warpem/warp/2.0.0dev29/download/linux-64/warp-2.0.0dev29-py311_0.tar.bz2`

No idea what is going on here, but the above works. Maybe something changed in the package configuration after dev29?

alisterburt, 2024-10-21:
strange, I can't reproduce either with conda or micromamba... glad you found a workaround though

```
$ conda create -n warptest warp=2.0.0 -c warpem -c nvidia/label/cuda-11.8.0 -c pytorch -c conda-forge
pytorch/linux-64 (check zst)                       Checked  0.1s
nvidia/label/cuda-11.8.0/noarch (check zst)         Checked  0.2s
nvidia/label/cuda-11.8.0/linux-64 (check zst)       Checked  0.2s
warpem/linux-64 (check zst)                         Checked  0.2s
pytorch/noarch (check zst)                          Checked  0.0s
nodefaults/noarch (check zst)                       Checked  0.2s
warpem/noarch (check zst)                           Checked  0.2s
nodefaults/linux-64 (check zst)                     Checked  0.1s
pytorch/linux-64                                   191.8kB @   3.6MB/s  0.1s
warpem/noarch                                      116.0 B @ 774.0 B/s  0.2s
nvidia/label/cuda-11.8.0/linux-64                             No change
nvidia/label/cuda-11.8.0/noarch                               No change
warpem/linux-64                                      2.6kB @  11.9kB/s  0.2s
pytorch/noarch                                                No change
nodefaults/noarch                                             No change
nodefaults/linux-64                                           No change
conda-forge/noarch                                  16.9MB @  30.8MB/s  0.4s
conda-forge/linux-64                                39.1MB @  52.2MB/s  0.7s

Transaction

  Prefix: /gpfs/scratchfs01/site/u/burta2/micromamba/envs/warptest

  Updating specs:

   - warp=2.0.0

  Package                          Version  Build                         Channel                        Size
───────────────────────────────────────────────────────────────────────────────────────────────────────────────
  Install:
───────────────────────────────────────────────────────────────────────────────────────────────────────────────

...
  + warp                        2.0.0dev29  py311_0                       warpem                        389MB
```

```
$ micromamba create -n warptest warp=2.0.0 -c warpem -c nvidia/label/cuda-11.8.0 -c pytorch -c conda-forge
warpem/noarch (check zst)                           Checked  0.2s
warpem/linux-64 (check zst)                         Checked  0.3s
pytorch/linux-64                                   191.8kB @   3.6MB/s  0.1s
nvidia/label/cuda-11.8.0/noarch                               No change
nvidia/label/cuda-11.8.0/linux-64                             No change
warpem/linux-64                                      2.6kB @  13.1kB/s  0.2s
pytorch/noarch                                                No change
warpem/noarch                                      116.0 B @ 402.0 B/s  0.3s
nodefaults/linux-64                                           No change
nodefaults/noarch                                             No change
conda-forge/noarch                                  16.9MB @  15.6MB/s  0.9s
conda-forge/linux-64                                39.1MB @  23.8MB/s  1.5s

Transaction

  Prefix: /apps/user/gRED/cryoem/prd/conda_envs/warptest

  Updating specs:

   - warp=2.0.0

  Package                          Version  Build                         Channel                        Size
───────────────────────────────────────────────────────────────────────────────────────────────────────────────
  Install:
───────────────────────────────────────────────────────────────────────────────────────────────────────────────

...
  + warp                        2.0.0dev29  py311_0                       warpem                        389MB

  Summary:

  Install: 181 packages

  Total download: 427MB

───────────────────────────────────────────────────────────────────────────────────────────────────────────────

rdrighetto, 2024-10-21:
I think our HPC may have some strange caching issues. At least there's a workaround. Feel free to close this issue if you find appropriate, thanks for looking into it!

closed 2024-10-21 by alisterburt