warpem/warp #279: update tilt series alignment wrapper to use AreTomo3 v2

alisterburt, 2024-10-23:
Might be good to update to AT3 given that AT2 is broken and their new release supports running on TS stacks like the old ones

> Dear Colleagues,
>
> We have released AreTomo3 version 2.0, which can be downloaded from [github.com/czimaginginstitute/AreTomo3](http://github.com/czimaginginstitute/AreTomo3). The new features and improvements are summarized below:
Offline batch processing can take MRC tilt series as input without mdoc files.
Runtime estimation of sample thickness.
More accurate CTF estimation with the implementation of John Heymann's spectral rescaling.
Estimation of defocus handedness.
Local CTF correction that takes into account two defocus gradients.
Hybrid CTF correction scheme. 
Composite metrics for comprehensive quality assessment
While some of these functions are still undergoing continuous improvement, we would like to share our most recent progress with you and are hoping they can be helpful in your journey towards many scientific breakthroughs.
>
>As always, your feedback is highly appreciated!
>
>Best,
Shawn

zxxshower, 2025-01-09:
A bit confused. The AT3 also has this problem by our test, v2.0.3. Therefore I guess we need to wait for a next upgrade? Still use the AT1 v1.3.4, which seems not that ideal...

alisterburt, 2025-01-09:
@zxxshower I assumed they would have fixed this by now but hadn't checked yet, good to know.

It's annoying to be 'behind' their  latest release but very little has changed that is relevant to us 🙂

zxxshower, 2025-01-10:
@alisterburt Thanks for your kind reply. Today I also tested the newest version v2.0.6, still not working.