warpem/warp #283: ts_etomo_patches fail: STAR file has no wrpDose or wrpTilt column

Phaips, 2024-10-28:
Hey,

I ran `WarpTools ts_etomo_patches --settings tiltseries.settings --patch_size 200 --angpix 10 --initial_axis 85` and it outputs:

```
Running command ts_etomo_patches with:
angpix = 10
mask = False
min_fov = 0
initial_axis = 85
do_axis_search = False
patch_size = 200
delete_intermediate = False
device_list = {  }
perdevice = 1
workers = {  }
settings = tiltseries.settings
input_data = {  }
input_data_recursive = False
input_processing = null
output_processing = null
input_norawdata = False

No alternative input specified, will use input parameters from tiltseries.settings
File search will be relative to /scicore/home/engel0006/vysamu91/warp/raw
2271 files found
Parsing previous results for each item, if available...
2271/2271, previous metadata found for 2271                                                                           
Connecting to workers...
Connected to 1 workers
Performing patch tracking alignment on all tilt-series...
```

and then it says:

```
Failed to process /scicore/home/engel0006/vysamu91/warp/tiltseries/Position_25_044_-34.00_20240710_054241_Fractions.tiff, marked as unselected
Check logs in /scicore/home/engel0006/vysamu91/warp/tiltseries/logs for more info.
Use the change_selection WarpTool to reactivate this item if required.
```

which fails for every item actually so 2271/2271 failed and in the logs it says:

```
2024-10-28 17:02:13.071 Received "TomoStack", with 2 arguments, for GPU #0, 12023 MB free:
2024-10-28 17:02:14.393 System.Exception: STAR file has no wrpDose or wrpTilt column.
2024-10-28 17:02:14.393    at Warp.TiltSeries.InitializeFromTomoStar(Star table) in /usr/share/miniconda/envs/package-build/conda-bld/warp_1728495487215/work/WarpLib/TiltSeries.cs:line 293
2024-10-28 17:02:14.393    at Warp.TiltSeries..ctor(String path, String dataDirectoryName) in /usr/share/miniconda/envs/package-build/conda-bld/warp_1728495487215/work/WarpLib/TiltSeries.cs:line 286
2024-10-28 17:02:14.393    at WarpWorker.WarpWorker.EvaluateCommand(NamedSerializableObject Command) in /usr/share/miniconda/envs/package-build/conda-bld/warp_1728495487215/work/WarpWorker/WarpWorker.cs:line 335
2024-10-28 17:02:23.959 10 seconds without heartbeat, exiting
```

Similarly, when trying to run `ts_stack` it also gives me the same error `STAR file has no wrpDose or wrpTilt column`
It might be a very silly error but I really don't seem to find it. My tomostar files have been sucessfully generated and don't seem to be wrong... one example `Position_8.tomostar`:

```
data_

loop_
_wrpMovieName #1
_wrpAngleTilt #2
_wrpAxisAngle #3
_wrpDose #4
_wrpAverageIntensity #5
_wrpMaskedFraction #6
  ../tiltseries/Position_8_056_-46.00_20240709_185616_Fractions.tiff  -46.00  -174.230  110  26.281  0.000
  ../tiltseries/Position_8_055_-44.00_20240709_185530_Fractions.tiff  -44.01  -174.230  108  26.625  0.000
```
I left out the rest of the tilts to not make this issue even longer but all the values seem fine to me.

On another note, when trying to run `WarpTools ts_aretomo --settings tiltseries.settings --angpix 10 --alignz 250` I get:

```
Unhandled exception. System.NullReferenceException: Object reference not set to an instance of an object.
   at WarpTools.Commands.AreTomoTiltseries.<>c.<Run>b__0_2(Movie m) in /usr/share/miniconda/envs/package-build/conda-bld/warp_1728495487215/work/WarpTools/Commands/Tiltseries/AreTomoTiltseries.cs:line 125
   at System.Linq.Enumerable.SelectArrayIterator`2.MoveNext()
   at Warp.Tools.MathHelper.Mean(IEnumerable`1 data) in /usr/share/miniconda/envs/package-build/conda-bld/warp_1728495487215/work/WarpLib/Tools/MathHelper.cs:line 40
   at Warp.Tools.MathHelper.GeometricMedian(IEnumerable`1 points, Single tolerance, Int32 maxIterations) in /usr/share/miniconda/envs/package-build/conda-bld/warp_1728495487215/work/WarpLib/Tools/MathHelper.cs:line 82
   at WarpTools.Commands.AreTomoTiltseries.<>c__DisplayClass0_0.<Run>b__0() in /usr/share/miniconda/envs/package-build/conda-bld/warp_1728495487215/work/WarpTools/Commands/Tiltseries/AreTomoTiltseries.cs:line 123
   at WarpTools.Commands.AreTomoTiltseries.Run(Object options) in /usr/share/miniconda/envs/package-build/conda-bld/warp_1728495487215/work/WarpTools/Commands/Tiltseries/AreTomoTiltseries.cs:line 138
   at WarpTools.WarpTools.Run(Object options) in /usr/share/miniconda/envs/package-build/conda-bld/warp_1728495487215/work/WarpTools/Program.cs:line 30
   at Warp.Tools.CommandLineParserHelper.ParseAndRun(String[] args, Func`2 run, Type[] verbs, String appName) in /usr/share/miniconda/envs/package-build/conda-bld/warp_1728495487215/work/WarpLib/Tools/CommandLineParserHelper.cs:line 26
   at WarpTools.WarpTools.Main(String[] args) in /usr/share/miniconda/envs/package-build/conda-bld/warp_1728495487215/work/WarpTools/Program.cs:line 17
   at WarpTools.WarpTools.<Main>(String[] args)
Aborted
```

which seems to be some kind of different error. Here, there is no logs file generated.

Any suggestions? Many thanks for the help!

alisterburt, 2024-10-28:
Hey @Phaips - it seems like your tiltseries.settings is not pointing at your tomostar files and is instead looking at your tiff files, try regenerating it with `create_settings` adding `--folder_data /path/to/tomostar/files` and `--extension "*.tomostar"`

Phaips, 2024-10-28:
Indeed a silly error from my side. Thank you @alisterburt for the quick reply ! Now I am just wondering why pointing at the raw .tiff files (or also .eer I guess) instead of the .tomostar in the beginning when the tomostar get created later. My initial settings were:
```
WarpTools create_settings \
-o tiltseries.settings \
--folder_processing tiltseries \
--folder_data raw \
--extension *.tiff \
--angpix 1.98 \
--gain_path 20240709_100119_EER_GainReference.gain \
--exposure 2 \
--tomo_dimensions 4096x4096x1280
```

closed 2024-10-28 by Phaips