warpem/warp #284: `MCore` scores all zero during refinement

alisterburt, 2024-10-29:
From Simon Jenni on the google group:

```quote
Running MCore e.g. like this:

MCore \
--population m_run01/population_01.population \
--refine_imagewarp 4x4 \
--refine_particles \
--refine_stageangles \
--refine_volumewarp 4x4x2x10 \
--ctf_defocus \
--perdevice_refine 1

Randomly, as it appears, all the refinement scores are 0 (see log file for one tiltseries below) and nothing is refined. This is after previous refinement runs of the same population/source/species were just fine. The program runs to the end and without generating any error. Has anyone observed also observed this?

All the best, Simon

m_run01/refinement_temp/logs/refine_Position_1_2.log

2024-10-29 19:10:23.049 Received "MPARefine", with 4 arguments, for GPU #0, 80067 MB free:
2024-10-29 19:10:23.094 Loading tilt series and masks...
2024-10-29 19:10:30.366 77439 MB after loading raw data of Position_1_2.tomostar
2024-10-29 19:10:30.372 Calculating spectral weights...
2024-10-29 19:10:32.464 75687 MB after spectra estimation of Position_1_2.tomostar
2024-10-29 19:10:32.471 Precomputing gradient weights...
2024-10-29 19:10:34.047 Re-extracting particles for optimization iteration 1/3
2024-10-29 19:10:34.179 Running optimization iteration 1/3, image & volume warping0
0

2024-10-29 19:10:34.685 Running optimization iteration 1/3, stage orientation0
0

2024-10-29 19:10:35.375 Running optimization iteration 1/3, particle poses0
0

2024-10-29 19:10:36.547 Running optimization iteration 1/3, CTF parameters0
0

2024-10-29 19:10:38.228 Re-extracting particles for optimization iteration 2/3
2024-10-29 19:10:38.376 Running optimization iteration 2/3, image & volume warping0
0

2024-10-29 19:10:38.870 Running optimization iteration 2/3, stage orientation0
0

2024-10-29 19:10:39.566 Running optimization iteration 2/3, particle poses0
0

2024-10-29 19:10:40.731 Running optimization iteration 2/3, CTF parameters0
0

2024-10-29 19:10:42.395 Re-extracting particles for optimization iteration 3/3
2024-10-29 19:10:42.526 Running optimization iteration 3/3, image & volume warping0
0

2024-10-29 19:10:43.014 Running optimization iteration 3/3, stage orientation0
0

2024-10-29 19:10:43.682 Running optimization iteration 3/3, particle poses0
0

2024-10-29 19:10:44.927 Running optimization iteration 3/3, CTF parameters0
0

2024-10-29 19:10:46.622 Re-extracting particles after optimization iteration 3/3
2024-10-29 19:10:46.753 Final score: 0

2024-10-29 19:10:46.858 74411 MB after optimization of Position_1_2.tomostar
2024-10-29 19:10:46.953 Calculating FRC between projections and particles for weight optimization
2024-10-29 19:10:47.364 77437 MB after optimization teardown of Position_1_2.tomostar
2024-10-29 19:10:47.367 Extracting and back-projecting particles...
2024-10-29 19:10:53.888 77437 MB after backprojection of Position_1_2.tomostar
2024-10-29 19:10:54.213 80061 MB after full refinement of Position_1_2.tomostar
2024-10-29 19:10:54.320 Finished refining Position_1_2.tomostar
2024-10-29 19:10:54.323 Execution took 31.270 seconds
2024-10-29 19:10:54.326 
```

deadlift385, 2024-10-30:
I really don't know at the moment under which condition this occurs. One thing I noticed is that the pre-processing step took substantially longer (e.g. >20 min instead of about 5 min) when the refinement failed with 0 scores.

deadlift385, 2024-10-31:
This is caused once <Param Name="GlobalResolution" Value="3.323468" /> in the species_01.species file gets too good. Not sure why and whether it is a memory issue or not. But something fails in the preprocessing step. If I set the value to 3.40000 or so, MCore refines OK again with the expected scores. I will run a few more test. I could upload ~24GB self-contained directories where the error is reproduced.

alisterburt, 2024-10-31:
@deadlift385 this is interesting, thanks - I'll chat to Dimitry about it and get back to you

alisterburt, 2024-10-31:
@deadlift385 it's definitely a memory issue, apparently when GPU kernels fail to allocate memory they crash and output all zeros... I see you're already at Nyquist so don't think there's a quick fix

alisterburt, 2024-11-01:
actually - there is a `--cpu_memory` option, does this let you run your refinements?

alisterburt, 2024-11-01:
lowering `--cpu_batch` may also be useful

deadlift385, 2024-11-01:
Thanks, Alister!

--cpu_memory and/or --ctf_batch
does not help in this case, as I think they don't affect the preprocessing step. I just need to run it on a better GPU with more memory.

A similar thing happens when you run WarpTools ts_export_particles with too large boxes (or particle diameter): it runs without error or crash, but some or all of the output 3D particle maps are just 0 (empty). It would be helpful to have a warning or error in these situations, but I don't know how hard it is to implement this.

deadlift385, 2025-01-30:
Maybe the box size per se (and then not having enough memory) is not the problem that causes this. I do observe the same also for a smaller box on the same graphics card. I have not the suspicion that this happens when in the input species_01.species file

        <Param Name="DenoisedAtResolution" Value="4.8" />
and / or 
        <Param Name="GlobalResolution" Value="4.8" />

reaches Nyquist.

alisterburt, 2025-01-30:
It looks like something is missing from the end of your message - interesting observation!

nkjaerjensen, 2025-04-13:
Hi @alisterburt

I have managed to encounter this exact error recently. Identically to the description from Simon, everything runs fine for initial MCore runs, but starting a new run once the .species GlobalResolution becomes sufficiently good results in 0 scores in the logs. The error persists across different GPUs (4090s and RTX 5000s), and also regardless of the number of GPUs (1-4 attempted). Even using the exact same MCore parameters (so not adding any new parameters) reproduces the 0-scores if the GlobalResolution parameter has become 'too good', whereever that exactly is. Similarly for me, this has only been observed at resolutions where Nyquist resolution has been obtained in parts of the map.

Interestingly, I can also reproduce the workaround that the scores are no longer zero in subsequent runs if I artificially change the .species GlobalResolution parameter to an artificially larger value (in my case from 3.74 to 3.76).

Weirdly, if I obtain a better GlobalResolution result after artificially changing the GlobalResolution parameter to a larger value, I can still keep on refining with more MCore runs without issues or 0-scores! This is the case even if the GlobalResolution number reported is better than the problematic GlobalResolution value that resulted in 0-scores (so it doesn't seem to be an issue that appears above certain resolutions). I have a hard time understanding how this may be.

As a workaround, is it theoretically 'safe' to artificially change the GlobalResolution parameter value or may it impact the quality of the refinement/results downstream?

Best wishes,
//nkj

alisterburt, 2025-04-14:
@nkjaerjensen @deadlift385 thanks for the thorough troubleshooting - this is really useful. 

I don't have a solution and we're stretched thin here so can't promise I will be able to look into it.

> Weirdly, if I obtain a better GlobalResolution result after artificially changing the GlobalResolution parameter to a larger value, I can still keep on refining with more MCore runs without issues [...]

Super weird.

> As a workaround, is it theoretically 'safe' to artificially change the GlobalResolution parameter value or may it impact the quality of the refinement/results downstream?

If this works with only very minimal changes to the resolution parameter as you suggest I would assume it's safe and a good workaround :-)

dtegunov, 2025-04-14:
Hey @nkjaerjensen! The lack of correlation with the number of GPUs is expected, but how much memory do those cards have individually? If you monitor nvidia-smi, does it come close to using all memory at some point?

deadlift385, 2025-04-15:
As I mentioned above, for me this, is also happening when you reach (or get close to) Nyquist resolution in DenoisedAtResolution and/or GlobalResolution.

	<Param Name="DenoisedAtResolution" Value="4.698" />
	<Param Name="GlobalResolution" Value="4.750787" />

nkjaerjensen, 2025-04-15:
Hi @dtegunov! That's a fair point, the RTX 5000s have 32 GB of memory individually, and the 4090s have 24 GBs individually.

To test the memory usage: I just found some time to revert all the data back to this zero-score version, and the problem is reproducible with the reverted metadata. Monitoring nvidia-smi roughly once per second, it never comes close to using all the memory. For my data, a maximum usage of ~5GB of GPU memory was observed during pre-processing, and a maximum usage of ~6.1 GB of GPU memory was observed during the actual refinement of individual tilt series in the source.