warpem/warp #288: unable to extract using `ts_export_particles` from RELION 5 refined STAR file

alisterburt, 2024-11-12:
got a report saying...

> Currently, Warp can’t directly extract from a relion5 refined star file or the M particles file

I'd be surprised if this is true but need to try to reproduce...

rnwokonko, 2025-04-18:
Hey Alister, I believe I might be having a similar issue but I'm not sure what exactly is the cause. I have a relion5 formatted star file with coordinates generated using another template matching software. For whatever reason, when I attempt to use the export particle function Warp throws an error 'Unhandled exception. System.Exception: Couldn't find rlnCoordinateX column in input STAR file.' All of my coordinates are in 'rlnCenteredCoordinate(X/Y/Z)Angst' format instead. I can get it to work if I revert to the relion3 format. Not sure if I should open a separate issue, or if my issue is similar to the one you mention here.

alisterburt, 2025-04-18:
@rnwokonko thanks for the report - we don't support the centered coordinate format in Warp right now... we explicitly describe a format that we support in the link below. In cases like this (non standard Warp/M workflow) you would be expected to script around discrepancies yourself 

https://warpem.com/reference/warptools/tomogram_particle_files/

rnwokonko, 2025-04-19:
Appreciate the quick reply. Cheers.