warpem/warp #289: Particle series export fails when asking for only a single tilt

shahpnmlab, 2024-11-13:
Also, DT mentioned that —n_tilts 1 should technically work when exporting particles series data, but the fact it doesnt is a bug, so here’s the error message

WarpTools ts_export_particles --settings warp_tiltseries.settings --input_star picked.star --output_star test.star --output_angpix 0.7303 --box 200 --coords_angpix 2.00 --n_tilts 1 --diameter 130 --2d --input_data tomostar/TS_12.tomostar
Running command ts_export_particles with:
input_star = picked.star
input_directory = null
input_pattern = *.star
coords_angpix = 2
normalized_coords = False
output_star = test.star
output_angpix = 0.7303
box = 200
diameter = 130
relative_output_paths = False
2d = True
3d = False
dont_normalize_input = False
dont_normalize_3d = False
n_tilts = 1
max_missing_tilts = 5
device_list = {  }
perdevice = 1
workers = {  }
settings = warp_tiltseries.settings
input_data = { tomostar/TS_12.tomostar }
input_data_recursive = False
input_processing = null
output_processing = null
input_norawdata = False

Using alternative input specified by --input_data
File search will be relative to training/warp_apo
1 files found
Parsing previous results for each item, if available...
1/1, previous metadata found for 1
Found 4127 particles in 31 tilt series
Connecting to workers...
Connected to 1 workers
Failed to process /warp_tiltseries/TS_12.tomostar, marked as unselected
Check logs in /home/pranav/warp_apo/warp_tiltseries/ogs for more info.
Use the change_selection WarpTool to reactivate this item if required.
1/1, 1 failed, 00:00 remaining
Finished processing in 00:00:13
Unhandled exception. System.InvalidOperationException: No particles to write out. Check that rlnMicrographName entries match tomostar file for each tilt series.
   at WarpTools.Commands.ExportParticlesTiltseries.WriteOutputStarFile(Dictionary`2 perTiltSeriesTables, String particleStarPath, Int32 outputDimensionality, Int32 maxMissingTilts) in /WarpTools/Commands/Tiltseries/ExportParticlesTiltseries.cs:line 1048
   at WarpTools.Commands.ExportParticlesTiltseries.Run(Object options) in /WarpTools/Commands/Tiltseries/ExportParticlesTiltseries.cs:line 383
   at WarpTools.WarpTools.Run(Object options) in WarpTools/Program.cs:line 30
   at Warp.Tools.CommandLineParserHelper.ParseAndRun(String[] args, Func`2 run, Type[] verbs, String appName) in /home/pranav/Desktop/pranav/software/warp/WarpLib/Tools/CommandLineParserHelper.cs:line 26
   at WarpTools.WarpTools.Main(String[] args) in /WarpTools/Program.cs:line 17
   at WarpTools.WarpTools.<Main>(String[] args)
Aborted (core dumped)

shahpnmlab, 2024-11-13:
And if you are fixing that - can I also then add that the subsequent star file should be SPA compliant and not STA...

alisterburt, 2024-11-13:
@shahpnmlab thanks for the report - interesting point about the output file, is this strictly necessary? Running refinement on a single 2D image in RELION is, as I understand it, exactly equivalent to running SPA

shahpnmlab, 2024-11-13:
IDK… just pre-empting any potential problems in case something like this is not supported further downstream… if you dont foresee any issue, then no need to fix what aint broken…

shahpnmlab, 2024-12-11:
Also curious if this was sorted out?

alisterburt, 2024-12-11:
@shahpnmlab haven't had a chance to look at it, sorry!

shahpnmlab, 2025-04-20:
Fixed in [366](https://github.com/warpem/warp/commit/a7bf679ee5342c967273bf8247f6c39edea196d8). TODO: Output SPA style STAR file.

closed 2025-04-20 by shahpnmlab