warpem/warp #291: Question: Different reconstruction coordinate system between Warp 1.0.9 and WarpTools

bwmr, 2024-11-14:
Hi, 

I have a previous Warp/M project with a decent ribosome average, and I wanted to play around with WarpTools to see whether I can replicate it without having to switch back and forth between Linux and Windows. 

First of all, I love the WarpTools processing workflow, it's very fast and easy to use, thank you for this upgrade!

I noticed however that there is both a shift and a flip around Z between the reconstructions created by WarpTools vs. those created by Warp 1.0.9 (where the coordinate system matches imod's), even if the same metadata files were used for importing the tiltseries into Warp and WarpTools. 

Is there any way of retaining the old coordinate system or transforming coordinates between the two, or any flag I should use while importing alignments? 

Thank you,
Benedikt

PS: While I of course see the point you raise in #195, that picking on the pipeline-generated tomograms should be preferred, I was a bit surprised that there's such a big difference between Warp 1.X and WarpTools.

alisterburt, 2024-11-14:
Hey @bwmr 

There's a small detail here which only comes up when you are reprocessing already processed data
- In Warp 1.X tilt angles were (optionally, but by default) flipped during import of IMOD alignments to maintain handedness
- in Warp 2.0 tilt angles are (optionally, but by default) flipped during import of the tilt series level metadata

As a consequence, Warp 2.0 expects that the tilt angles were already flipped during import and doesn't re-flip when importing from alignments. Does this make sense?

Re: the shift, the origin of the tomogram coordinate system is (and always has been...) defined by the combination of
- shifts in the xf file
- the tomogram dimensions, defined in unbinned pixels 

Tomogram dimensions used to be set in the GUI and are now set during settings creation - I expect a difference here is the source of your observed shift

Thanks for the kind words and happy processing!

Cheers,

Alister

bwmr, 2024-11-18:
Hey @alisterburt , 

Thanks for the quick response and the detailed explanation! 

- Running `WarpTools ts_reconstruct` on the Warp 1.X-generated tomostar files gives tomograms with the right orientation and coordinate system.
- The WarpTools 2.X `ts_import` generates tomostar-files with correctly flipped angles from the mdoc file.
- When I import previous alignments using `ts_import_alignments`, and I manually invert the tilt angles in the `.tlt` file, this indeed fixed the issue - the reconstruction now matches that of Warp 1.X and imod. 

Thanks again for your help. 

Cheers,
Benedikt

closed 2024-11-18 by bwmr

alisterburt, 2024-11-18:
@bwmr thanks for following up and glad you got sorted! Happy processing 🙂