warpem/warp #294: mrc file header error after fs_motion_and_ctf

frozenmess, 2024-11-19:
After running the fs_motion_and_ctf job I see that the subsequent mrc files in /frameseries/average/ look strange/unexpected (based on other processed tilt series from another dataset) in 3dmod, but also do not have any header information. I noticed this after having errors downstream with WarpTools ts_stack. Here is the full error when I check the mrc header:

ERROR: mrc_head_read - bad file mode 12.
ERROR: iiOpen - tilt_-36.0.mrc has unknown format.

ERROR: iiuOpen - Could not open 'tilt_-36.0.mrc'

This has been replicated while testing multiple releases (dev29, dev28, dev22, and dev10). Motion correction and ctf estimation seem to work fine as assessed by histograms from WarpTools filter_quality. I installed warp using mamba. Both the frame.settings and tilt.settings files seem to contain the correct information. Any help is much appreciated. Thank you.

alisterburt, 2024-11-19:
Hi @frozenmess - data are stored in float16 to save disk space. You are using an older version of IMOD which cannot read MRC files with data stored in float16. You should upgrade your IMOD installation 🙂 

If you really need compatibility you can force Warp to output float32 data https://warpem.github.io/warp/reference/warptools/environment_variables/#force-writing-32-bit-mrcs

frozenmess, 2024-11-19:
After updating to IMOD 4.11.25 (previously had 4.11.4) I am still having the same issues with the mrc files.

alisterburt, 2024-11-19:
https://bio3d.colorado.edu/ftp/latestIMOD/