warpem/warp #296: issue with fs_motion_and_ctf  (error: System.OverflowException: Arithmetic operation resulted in an overflow.)

zxxshower, 2024-11-20:
Hi,
We just encontered the error with fs_motion_and_ctf function.
Here is what we can read from the report in terminal:

Connecting to workers...
Connected to 5 workers
Failed to process /fs/folder_path/warp_frameseries/Position_2_2_045_29.00_20240817_201823_EER.eer, marked as unselected
Check logs in /fs/folder_path/warp_frameseries/logs for more info.
Use the change_selection WarpTool to reactivate this item if required.

and the log file writes:

2024-11-20 10:58:38.204 Received "LoadStack", with 4 arguments, for GPU #6, 23723 MB free:
2024-11-20 10:58:42.316 Loaded stack: 4096, 4096, 31, normal, real, 1 A/px, ID = -1, 1
2024-11-20 10:58:42.317 Execution took 4.112 seconds
2024-11-20 10:58:42.317
2024-11-20 10:58:42.319 Received "MovieProcessMovement", with 2 arguments, for GPU #6, 23723 MB free:
2024-11-20 10:58:43.347 Processed movement for /fs/folder_path/warp_frameseries/Position_2_2_045_29.00_20240817_201823_EER.eer
2024-11-20 10:58:43.347 Execution took 1.027 seconds
2024-11-20 10:58:43.348
2024-11-20 10:58:43.350 Received "MovieExportMovie", with 2 arguments, for GPU #6, 23723 MB free:
2024-11-20 10:58:44.904 0: 804
2024-11-20 10:58:44.904 1: 5
2024-11-20 10:58:44.919 2: 20
2024-11-20 10:58:44.919 3: 1
2024-11-20 10:58:44.920 4: 759
2024-11-20 10:58:44.920 5: 0
2024-11-20 10:58:44.920 6: 0
2024-11-20 10:58:44.921 7: 6
2024-11-20 10:58:44.941 Exported movie for /fs/folder_path/warp_frameseries/Position_2_2_045_29.00_20240817_201823_EER.eer
2024-11-20 10:58:44.941 Execution took 1.580 seconds
2024-11-20 10:58:44.946
2024-11-20 10:58:44.949 Received "LoadStack", with 4 arguments, for GPU #6, 23723 MB free:
2024-11-20 10:58:45.051 Loaded stack: 4096, 4096, 1, normal, real, 1 A/px, ID = -1, 1
2024-11-20 10:58:45.100 Execution took 0.151 seconds
2024-11-20 10:58:45.119
2024-11-20 10:58:45.146 Received "MovieProcessCTF", with 2 arguments, for GPU #6, 23723 MB free:
2024-11-20 10:58:47.765 System.OverflowException: Arithmetic operation resulted in an overflow.
2024-11-20 10:58:47.765    at Warp.Cubic1D..ctor(float2[] data) in /usr/share/miniconda/envs/package-build/conda-bld/warp_1728495487215/work/WarpLib/Cubic1D.cs:line 30
2024-11-20 10:58:47.765    at Warp.Cubic1D.FitCTF(float2[] data, Single[] simulation, Single[] zeros, Single[] peaks, Cubic1D& background, Cubic1D& scale) in /usr/share/miniconda/envs/package-build/conda-bld/warp_1728495487215/work/WarpLib/Cubic1D.cs:line 215
2024-11-20 10:58:47.765    at Warp.Movie.ProcessCTF(Image originalStack, ProcessingOptionsMovieCTF options) in /usr/share/miniconda/envs/package-build/conda-bld/warp_1728495487215/work/WarpLib/Movie.cs:line 1941
2024-11-20 10:58:47.765    at WarpWorker.WarpWorker.EvaluateCommand(NamedSerializableObject Command) in /usr/share/miniconda/envs/package-build/conda-bld/warp_1728495487215/work/WarpWorker/WarpWorker.cs:line 265

I don't have any idea why this happens, this eer file seems normal, and there is no error with this eer file during MotionCor + CtfFind in relion5.
Thanks!

zxxshower, 2024-11-20:
By the way, all the rest eer files (900/901) finished without any error

zxxshower, 2024-11-21:
@alisterburt Thanks for your reply, you are right, presumably the CTF fitting procedure has some problem.
I changed the --c_grid value from 2x2x1 to 5x5x1, and increased slightly --c_range_max from 8 to 9 (pix: 3.03 Å), then it went well. Maybe the latter parameter needs some test for getting an optimal value in this case. I'll close this issue.

closed 2024-11-21 by zxxshower

ThatIsMyUsername, 2025-02-18:
> [@alisterburt](https://github.com/alisterburt) Thanks for your reply, you are right, presumably the CTF fitting procedure has some problem. I changed the --c_grid value from 2x2x1 to 5x5x1, and [...]

Hi, you mentioned that changing the --c_grid value from 2x2x1 to 5x5x1, and changing --c_range_max from 8 to 9 successfully solved this problem. I want to know why this modification can solve the problem? In other words, why 5x5x1 and 9, and not some other value, what is the specific reason for the change.
Thank you.

zxxshower, 2025-02-18:
The c_grid value is not the critical parameter in this case. As I mentioned that the latter parameter (c_range_max) require some optimization. 
I don't know the specific reason, pure assumption: poor (or incorrect) high-resolution signals may lead to incorrect CTF fitting in Warp. To address this issue, cutting off more of the high-resolution signal during CTF fitting might help. 
You could also try adjusting other parameters, such as c_window and c_range_min, using the file that reported the error.

ThatIsMyUsername, 2025-02-18:
> The c_grid value is not the critical parameter in this case. As I mentioned that the latter parameter (c_range_max) require some optimization. I don't know the specific reason, pure assumption: poor [...]

Thank you very much for your reply, which successfully helped me solve the problem. The parameter **--c_range_max** is really an important parameter.
Thanks again.