warpem/warp #308: Enforce naming conventions when importing tilt series (or at any point earlier than template matching).

uermel, 2024-11-27:
Using the command line workflow, I ran into this error when I wanted to threshold template matching results: 
`Unhandled exception. System.Exception: Please provide a suffix with an exact match for 24aug01b_Position_20_{in_suffix}.star`

This seems to occur for me because 2 tilt series have overlapping names: 
* 24aug01b_Position_20
* 24aug01b_Position_20_3

These can't be differentiated by a simple glob pattern. It's a reasonable requirement for names to be uniquely globbable, but it would be nice if this was enforced at an earlier point in the pipeline, rather than throwing an error only at the template matching step. 

Unfortunately, these types of names are very common for tilt series coming out of Tomo5 (😭), so I'd expect this to become a common issue if users don't change the tilt series names.

dtegunov, 2024-11-27:
Ah, one user at the EMBO workshop ran into this, but I thought it was a uniquely unfortunate case of naming. Too bad Tomo5 makes it standard. I'll check if it can be fixed easily. Before I go looking through all Wools, this only occurred in threshold_picks, right?

uermel, 2024-11-27:
Yes, this was only an issue with `threshold_picks`, everything up to that point worked fine.

alisterburt, 2025-04-28:
this should be solved already by this little block... if you provide exactly the right suffix things should work out

https://github.com/warpem/warp/blob/7c5e5c622d9e97c91da558d88bb2b9f341c3bf39/WarpTools/Commands/General/ThresholdPicks.cs#L82-L86

will close for now but please reopen if it's still an issue

closed 2025-04-28 by alisterburt