warpem/warp #310: ts_import error of "No time stamp found"

CleoShen, 2024-12-03:
Hi there, I'm running the "WarpTools ts_import". And returns error with something like: **`No images left for 20241030_grid2_WTISV_p10_ts_004.mrc.mdoc after removing tilts based on selection status, max tilt angle and masked fraction criteria
Failed to parse 20241030_grid2_WTISV_p06_ts_009.mrc.mdoc: No time stamp found: 20241030_grid2_WTISV_p06_ts_009_20241030_grid2_WTISV_p06_ts_009_00001_-66.0.mrc 
Failed to parse 20241030_grid2_WTISV_p03_ts_009.mrc.mdoc: No time stamp found: 20241030_grid2_WTISV_p03_ts_009_20241030_grid2_WTISV_p03_ts_009_00001_-66.0.mrc 
Failed to parse 20241030_grid2_WTISV_p03_ts_005.mrc.mdoc: No time stamp found: 20241030_grid2_WTISV_p03_ts_005_20241030_grid2_WTISV_p03_ts_005_00001_-66.0.mrc 
Failed to parse 20241030_grid2_WTISV_p07_ts_003.mrc.mdoc: No time stamp found: 20241030_grid2_WTISV_p07_ts_003_20241030_grid2_WTISV_p07_ts_003_00001_-66.0.mrc 
Failed to parse 20241030_grid2_WTISV_p05_ts_007.mrc.mdoc: No time stamp found: 20241030_grid2_WTISV_p05_ts_007_20241030_grid2_WTISV_p05_ts_007_00001_-66.0.mrc 
Failed to parse 20241030_grid2_WTISV_p05_ts_005.mrc.mdoc: No time stamp found: 20241030_grid2_WTISV_p05_ts_005_20241030_grid2_WTISV_p05_ts_005_00001_-66.0.mrc 
Failed to parse 20241030_grid2_WTISV_p07_ts_002.mrc.mdoc: No time stamp found: 20241030_grid2_WTISV_p07_ts_002_20241030_grid2_WTISV_p07_ts_002_00001_-66.0.mrc 
Successfully parsed 0 MDOC files, 61 failed`**

I've tried to replace the year -2024 to -24 in the DataTime line, but still returns this error.

alisterburt, 2024-12-03:
Hi @CleoShen - do your mdocs differ significantly from those in the tutorial? How were your data collected? Please paste the contents of an mdoc here, I can't do much to help troubleshoot without it...

CleoShen, 2024-12-03:
[20241030_grid2_WTISV_p03_ts_002.txt](https://github.com/user-attachments/files/17987750/20241030_grid2_WTISV_p03_ts_002.txt)
Hi Alis, I uploaded one mdoc changing the suffix .mrc.mdoc with .txt for permission. The content should be the same. FYI.

alisterburt, 2024-12-03:
@CleoShen it appears most of your mdocs are for individual files rather than tilt series and those are the ones that have failed with the timestamp error

CleoShen, 2024-12-03:
Thank you for your fast responses.
I don’t understand what you mean about “for individual files”? I tested these mdocs by using standalone DriftCombine, the TS can be created by using these mdocs.

alisterburt, 2024-12-03:
Apologies, I misread your error. Here is the relevant code block:
https://github.com/warpem/warp/blob/712c02e0579816fc8575371238da4b27fb0845c9/WarpTools/Commands/Tiltseries/ImportTiltseries.cs#L208-L234

You're hitting an error which shows that no timestamp was found - this most likely means that you are missing a `DateTime` entry in relevant mdoc sections for those tilt series which show errors.

note: I do not see an error for the mdoc that you shared

CleoShen, 2024-12-03:
No problem. So if the mdoc is fine, will the error relate with my tilt micrographs?

https://github.com/warpem/warp/blob/712c02e0579816fc8575371238da4b27fb0845c9/WarpTools/Commands/Tiltseries/ImportTiltseries.cs#L208-L234

alisterburt, 2024-12-03:
the error implies the mdoc is not fine

alisterburt, 2024-12-03:
please compare your mdocs to those provided in the tutorial, carefully

CleoShen, 2024-12-03:
I just ran a successful job by using unsorted Mdoc files. Basically, I used SerialEM for data collection and it automatically generates sorted mdoc files listing the tilt frames from small to large angles. These sorted mdoc files would fail the ts_improt job, but the unsorted mdoc files listing does symmetry sequence tilt frames will not fail the job. FYI.

alisterburt, 2024-12-03:
Are you collecting with PACE tomo? If so I think it is writing out mdocs which look different

CleoShen, 2024-12-03:
Yes, I used PACEtomo script.