warpem/warp #314: Citation question: Any additional citation e.g. to the github repo when using warp2.0?

rkjensen, 2024-12-06:
Quick question: If I want to cite the use of warp and M I of course cite the two original papers, and write which version I used, but are there any specific way you want us to further cite the use of Warp2.0? E.g. with link to github repo or similar.

If you land on something specific, maybe you could add it to the README and https://warpem.github.io.

alisterburt, 2024-12-06:
Hi Rasmus,

Thanks for asking about this - here is a DOI you can cite 

https://zenodo.org/records/14218750

It is in a badge at the top of the readme but is probably not prominent enough, will add a note as you suggest!

rkjensen, 2024-12-06:
Ahh yes I see it now, but missed it before so it would maybe be good to make it more prominent!

closed 2024-12-06 by rkjensen