warpem/warp #315: Breaking changes to particle export?

shahpnmlab, 2024-12-09:
Hi Guys,
Has something changed in particles_export routine in WarpTools? I am on commit 95c6382. I have previously successfully extracted subtomos from this star file and this dataset (in fact I am currently processing this data as we speak). When i try to extract particles I am seeing the following error message `Unhandled exception. System.InvalidOperationException: No particles to write out. Check that rlnMicrographName entries match tomostar file for each tilt series.`
Below is my command and full output
```
WarpTools ts_export_particles --settings tiltseries.settings --coords_angpix 5.8  --input_star tomopicker_coords.star --output_star test/star.star --box 32 --diameter 130 --3d --output_angpix 10 --input_data tomostar/gs01_ts_003.tomostar --output_processing test/ --relative_output_paths
Running command ts_export_particles with:
input_star = tomopicker_coords.star
input_directory = null
input_pattern = *.star
coords_angpix = 5.8
normalized_coords = False
output_star = test/star.star
output_angpix = 10
box = 32
diameter = 130
relative_output_paths = True
2d = False
3d = True
dont_normalize_input = False
dont_normalize_3d = False
n_tilts = null
max_missing_tilts = 5
device_list = {  }
perdevice = 1
workers = {  }
settings = tiltseries.settings
input_data = { tomostar/gs01_ts_003.tomostar }
input_data_recursive = False
input_processing = null
output_processing = test/
input_norawdata = False

Using alternative input specified by --input_data
File search will be relative to /gpfs3/well/stuart/users/dce358/20240914_1_NSP2_6_NSP5_AuFid_pnms_stuart
1 files found
Parsing previous results for each item, if available...
1/1, previous metadata found for 1
Found 77675 particles in 41 tilt series
Connecting to workers...
Connected to 1 workers
Failed to process /gpfs3/well/stuart/users/dce358/20240914_1_NSP2_6_NSP5_AuFid_pnms_stuart/test/gs01_ts_003.tomostar, marked as unselected
Check logs in /gpfs3/well/stuart/users/dce358/20240914_1_NSP2_6_NSP5_AuFid_pnms_stuart/test/logs for more info.
Use the change_selection WarpTool to reactivate this item if required.
1/1, 1 failed, 00:00 remaining
Finished processing in 00:00:01
Unhandled exception. System.InvalidOperationException: No particles to write out. Check that rlnMicrographName entries match tomostar file for each tilt series.
   at WarpTools.Commands.ExportParticlesTiltseries.WriteOutputStarFile(Dictionary`2 perTiltSeriesTables, String particleStarPath, Int32 outputDimensionality, Int32 maxMissingTilts) in /exafs1/well/strubi-stuart/users/dce358/softwares_for_all/warp/WarpTools/Commands/Tiltseries/ExportParticlesTiltseries.cs:line 1048
   at WarpTools.Commands.ExportParticlesTiltseries.Run(Object options) in /exafs1/well/strubi-stuart/users/dce358/softwares_for_all/warp/WarpTools/Commands/Tiltseries/ExportParticlesTiltseries.cs:line 383
   at WarpTools.WarpTools.Run(Object options) in /exafs1/well/strubi-stuart/users/dce358/softwares_for_all/warp/WarpTools/Program.cs:line 30
   at Warp.Tools.CommandLineParserHelper.ParseAndRun(String[] args, Func`2 run, Type[] verbs, String appName) in /exafs1/well/strubi-stuart/users/dce358/softwares_for_all/warp/WarpLib/Tools/CommandLineParserHelper.cs:line 26
   at WarpTools.WarpTools.Main(String[] args) in /exafs1/well/strubi-stuart/users/dce358/softwares_for_all/warp/WarpTools/Program.cs:line 17
   at WarpTools.WarpTools.<Main>(String[] args)
Aborted (core dumped)
```

And here are the first 15 lines of the star file
```
# Created by the starfile Python package (version 0.4.11) at 20:12:10 on 26/11/2024

data_particles

loop_
_rlnMicrographName #1
_rlnCoordinateX #2
_rlnCoordinateY #3
_rlnCoordinateZ #4
_rlnAutopickFigureOfMerit #5
gs01_ts_003.tomostar    191.00  372.00  83.00   0.52
gs01_ts_003.tomostar    413.00  264.00  281.00  0.52
gs01_ts_003.tomostar    483.00  155.00  288.00  0.50
gs01_ts_003.tomostar    440.00  222.00  55.00   0.50
gs01_ts_003.tomostar    38.00   132.00  222.00  0.50
gs01_ts_003.tomostar    208.00  143.00  163.00  0.50
gs01_ts_003.tomostar    251.00  466.00  55.00   0.50
gs01_ts_003.tomostar    317.00  373.00  134.00  0.49
gs01_ts_003.tomostar    184.00  109.00  173.00  0.49
gs01_ts_003.tomostar    286.00  185.00  219.00  0.49
gs01_ts_003.tomostar    465.00  259.00  44.00   0.49
gs01_ts_003.tomostar    327.00  381.00  220.00  0.48
gs01_ts_003.tomostar    338.00  393.00  280.00  0.48
gs01_ts_003.tomostar    432.00  240.00  48.00   0.48
gs01_ts_003.tomostar    187.00  140.00  84.00   0.48
```

shahpnmlab, 2024-12-09:
Ok, heres the output with WARP_DEBUG=1
```
cat -p worker0.out
Running on GPU #0 (45192 MB free), port 41085

Received "LoadGainRef", with 5 arguments, for GPU #0, 45192 MB free:
Loaded gain reference and defect map: , False, False, False,
Execution took 0.000 seconds

fail: Microsoft.AspNetCore.Server.Kestrel[13]
      Connection id "0HN8OEA84DNIE", Request id "0HN8OEA84DNIE:00000001": An unhandled exception was thrown by the application.
      Microsoft.AspNetCore.Server.Kestrel.Core.BadHttpRequestException: Request body too large. The max request body size is 30000000 bytes.
         at Microsoft.AspNetCore.Server.Kestrel.Core.Internal.Http.Http1ContentLengthMessageBody.OnReadStarting()
         at Microsoft.AspNetCore.Server.Kestrel.Core.Internal.Http.MessageBody.TryStartAsync()
         at Microsoft.AspNetCore.Server.Kestrel.Core.Internal.Http.Http1ContentLengthMessageBody.ReadAsyncInternal(CancellationToken cancellationToken)
         at System.Runtime.CompilerServices.PoolingAsyncValueTaskMethodBuilder`1.StateMachineBox`1.System.Threading.Tasks.Sources.IValueTaskSource<TResult>.GetResult(Int16 token)
         at Microsoft.AspNetCore.Server.Kestrel.Core.Internal.Http.HttpRequestStream.ReadAsyncInternal(Memory`1 destination, CancellationToken cancellationToken)
         at System.Runtime.CompilerServices.PoolingAsyncValueTaskMethodBuilder`1.StateMachineBox`1.System.Threading.Tasks.Sources.IValueTaskSource<TResult>.GetResult(Int16 token)
         at System.IO.StreamReader.ReadBufferAsync(CancellationToken cancellationToken)
         at System.IO.StreamReader.ReadToEndAsyncInternal(CancellationToken cancellationToken)
         at WarpWorker.API.ServiceController.EvaluateCommand() in /exafs1/well/strubi-stuart/users/dce358/softwares_for_all/warp/WarpWorker/API/Service.cs:line 40
         at lambda_method4(Closure, Object, Object[])
         at Microsoft.AspNetCore.Mvc.Infrastructure.ActionMethodExecutor.SyncActionResultExecutor.Execute(ActionContext actionContext, IActionResultTypeMapper mapper, ObjectMethodExecutor executor, Object controller, Object[] arguments)
         at Microsoft.AspNetCore.Mvc.Infrastructure.ControllerActionInvoker.InvokeActionMethodAsync()
         at Microsoft.AspNetCore.Mvc.Infrastructure.ControllerActionInvoker.Next(State& next, Scope& scope, Object& state, Boolean& isCompleted)
         at Microsoft.AspNetCore.Mvc.Infrastructure.ControllerActionInvoker.InvokeNextActionFilterAsync()
      --- End of stack trace from previous location ---
         at Microsoft.AspNetCore.Mvc.Infrastructure.ControllerActionInvoker.Rethrow(ActionExecutedContextSealed context)
         at Microsoft.AspNetCore.Mvc.Infrastructure.ControllerActionInvoker.Next(State& next, Scope& scope, Object& state, Boolean& isCompleted)
         at Microsoft.AspNetCore.Mvc.Infrastructure.ControllerActionInvoker.InvokeInnerFilterAsync()
      --- End of stack trace from previous location ---
         at Microsoft.AspNetCore.Mvc.Infrastructure.ResourceInvoker.<InvokeFilterPipelineAsync>g__Awaited|20_0(ResourceInvoker invoker, Task lastTask, State next, Scope scope, Object state, Boolean isCompleted)
         at Microsoft.AspNetCore.Mvc.Infrastructure.ResourceInvoker.<InvokeAsync>g__Awaited|17_0(ResourceInvoker invoker, Task task, IDisposable scope)
         at Microsoft.AspNetCore.Mvc.Infrastructure.ResourceInvoker.<InvokeAsync>g__Awaited|17_0(ResourceInvoker invoker, Task task, IDisposable scope)
         at Swashbuckle.AspNetCore.SwaggerUI.SwaggerUIMiddleware.Invoke(HttpContext httpContext)
         at Swashbuckle.AspNetCore.Swagger.SwaggerMiddleware.Invoke(HttpContext httpContext, ISwaggerProvider swaggerProvider)
         at Microsoft.AspNetCore.Server.Kestrel.Core.Internal.Http.HttpProtocol.ProcessRequests[TContext](IHttpApplication`1 application)
10 seconds without heartbeat, exiting
```

alisterburt, 2024-12-09:
@shahpnmlab hmm, nearly 2k particles per tomogram might make the request too large -> if you reduce the number of particles per tomogram to 100 does it then run through?

shahpnmlab, 2024-12-09:
These are oversampled points, which is why there are so many... I will ask my colleague to try with a subset.

shahpnmlab, 2024-12-09:
@alisterburt ok, I can confirm that running with 100 particles only resulted in a successful extraction. Can the buffer size be adaptively increased by any chance?

alisterburt, 2024-12-09:
@shahpnmlab pushed a fix - you're compiling yourself so I don't need to cut a release, right?

>30MB for the request seems very large, is there one tomogram which has a really excessive number of particles? Are your ts xml files crazy large for some reason?

closed 2024-12-09 by alisterburt

shahpnmlab, 2024-12-09:
Hero! Yeah, I am compiling it myself atm