warpem/warp #317: Tilt series alignment - could not find *.xf

rnwokonko, 2024-12-09:
I am trying to process an EER tilt series data set collected w/ Tomo5 (Falcon4i), and am struggling to push past the alignment steps trying both IMOD and Aretomo2. Specifically, during alignment with either IMOD or Aretomo, I am getting an error stating that 'Failed to import alignments: Could not find xxx.xf'. I am wondering how I should go about trouble shooting this issue to move on with processing? I am able to process this same data set with the Windows version of Warp (GUI version), but am struggling with this using WarpTools.

Appreciate any insight you could provide.

Best,
Robert N.

alisterburt, 2024-12-10:
Hi @rnwokonko - there are some issues with AreTomo2 specifically (#279, #159 and czimaginginstitute/AreTomo2#21) - please try again with AreTomo v1.3.4 from the UCSF page

You can also always import alignments not made using our wrappers as before https://warpem.github.io/warp/reference/warptools/custom_tilt_series_alignments/

closing here but do reopen/add another comment if anything feels unclear!

alisterburt, 2024-12-10:
also: did you look at the worker logs to see where exactly things were failing?

closed 2024-12-10 by alisterburt

rnwokonko, 2024-12-10:
Thanks for the quick response. I actually didn't think to check out the closed issues on here until after I submitted. My issue looks very similar to  closed issue https://github.com/warpem/warp/issues/255. I will import my AreTomo alignment files until I can figure out the issue. 

The common error in the log file (for etomo) is: 

2024-12-10 00:44:03.592 Running align.com ... Error executing align.com^G
2024-12-10 00:44:03.593 ERROR: TILTALIGN - Allocating memory in surfaceSort
2024-12-10 00:44:03.593 ERROR: tiltalign -StandardInput: exited with status 1

Right now I am running this on a single A100 with plenty of memory. I'll look into this more.

alisterburt, 2024-12-10:
No worries! Tiltalign is not GPU accelerated, it's your RAM that's limiting there