warpem/warp #318: Trying to build using Visual Studio in Windows - missing tiffio.h

JulianoFaria, 2024-12-10:
The library NativeAcceleration does not have a path or reference for this library. I believe it is due wrong include path C:\Users\burta2\AppData\Local\miniforge3\envs\py\Library\include
[image]

alisterburt, 2024-12-10:
Hi @JulianoFaria - great that you're trying to get a development setup going. libtiff is a dependenchy and we do not bundle it. On my windows dev setup I pull it in from conda-forge. 

I'm only minimally familiar with developing software in this C#/dotnet ecosystem and on Windows. As such, I don't actually know how to provide a good developer experience around this sort of thing or what expectations are, any advice would be appreciated if you feel like you know more 🙂

JulianoFaria, 2024-12-11:
Hi @alisterburt, since I’m working part-time on this, I just need a bit more time to summarize the details I encountered while trying to build the solution.

JulianoFaria, 2024-12-11:
Just 1 point, not to forget:
The file jpeg8.dll is missing in the NativeAcceleration.dll post build process:
[image]
It caused the program to crash failing to load, but the prompt msg was:
[image]
To find the missing reference, i used the windows program depends:
[image]

JulianoFaria, 2024-12-15:
Hi @alisterburt , I was able to build the solution, i'd like to share the steps and my view about some possible improvements:

1. Would be nice to have a session for windows build in the readme.md
2. The website for download was down, just saw after I build the release session. if you could redirect the link it would be nice.
3. The conda-forge is a good way to handle references, but it would be nice to:
    - have all links needed pointing to a single environment variable and them to the needed folder (add in the readme the variable description): $(CONDA_ENV)\Library\include instead of C:\Users\burta2\AppData\Local\miniforge3\envs\py\Library\include. The places i found it were in the solution Libtorchsharpe and NativeAcceleration and in the sessions:
          - C/C++ -> Aditional Include Directories 
          - Linkier -> input > Aditional dependencies
          - Build Events -> Post build events
4. The solution is ponted to use conda 11.6, if other version is used, it is needed to change the project file NativeAcceleration.vcxproj in two lines (  <ImportGroup Label="ExtensionSettings">
    <Import Project="$(VCTargetsPath)\BuildCustomizations\CUDA 11.6.props" /> and   <ImportGroup Label="ExtensionTargets">
    <Import Project="$(VCTargetsPath)\BuildCustomizations\CUDA 11.6.targets" />
  </ImportGroup>)
5. I tried also to use newer version, but it does not work in versions above 12.0 due to deprecated [texture](https://forums.developer.nvidia.com/t/cudabindtexture2d-other-tex-functions-marked-deprecated-in-toolkit-11/182481) object 
6. The project LibTorchSharp is pointing to the source inside the python wrapper. Conda provides [libtorch](https://prefix.dev/channels/conda-forge/packages/libtorch) (the native reference code for the pyhton wrapper pytorch.
7. It would be nice to add the version of the references in the conda yaml file.
8. In the windows publish file, there was no instruction to build de wrapper dlls (Libtorchsharpe and NativeAcceleration)

Thanks,

Juliano

alisterburt, 2024-12-16:
Hi @JulianoFaria - thanks, this summary is very useful! 

> The website for download was down

which website? warpem.com has been retired, we now live at https://warpem.github.io/warp

JulianoFaria, 2024-12-19:
Hi @alisterburt, just the reference in the readme for windows install warpem.com, it should point to the repository release link:

[image]

In the  https://warpem.github.io/warp, no session or reference to install.