warpem/warp #319: MCore IndexOutOf Range Error

shahpnmlab, 2024-12-11:
Reconstructing particles in MCore (prior to running a refinement) results in System.IndexOutOfRangeException. I am on commit `87245fa`. Below is the stack trace
```
MCore --population m_run2/px2_run2.population --perdevice_refine 1 --min_particles 250 --refine_imagewarp 2x2 --refine_particles
Loading population... Done
Creating directories... Done
Spawning workers... Done
Preparing for refinement – this will take a few minutes per species
Preparing refinement requisites...
2/2
Performing refinement
Preparing population for data source px2_run2...Done
Loading gain reference for px2_run2... Done
Refining all series in data source...
30/41Unhandled exception. System.Exception: BadRequest: System.Net.Http.HttpConnectionResponseContent
   at Warp.WorkerWrapper.SendCommand(NamedSerializableObject command) in /home/pranav/Desktop/pranav/software/warp/WarpLib/WorkerWrapper.cs:line 231
   at Warp.WorkerWrapper.MPARefine(String path, String workingDirectory, ProcessingOptionsMPARefine options, DataSource source) in /home/pranav/Desktop/pranav/software/warp/WarpLib/WorkerWrapper.cs:line 458
   at MCore.MCore.<>c__DisplayClass16_4.<DoProcessing>b__4(Int32 ifile, Int32 threadID) in /home/pranav/Desktop/pranav/software/warp/MCore/MCore.cs:line 473
   at Warp.Tools.Helper.<>c__DisplayClass51_0.<ForCPU>b__0(Object id) in /home/pranav/Desktop/pranav/software/warp/WarpLib/Tools/Helper.cs:line 719
 Aborted (core dumped)
  ```
   
 output of worker_crash -
 ```
 System.IndexOutOfRangeException: Index was outside the bounds of the array.
   at Warp.TiltSeries.<>c__DisplayClass100_0.<PerformMultiParticleRefinement>b__19(Int32 t) in /home/pranav/Desktop/pranav/software/warp/WarpLib/TiltSeries.cs:line 6382
   at Warp.Tools.Helper.ArrayOfFunction[T](Func`2 generator, Int32 n) in /home/pranav/Desktop/pranav/software/warp/WarpLib/Tools/Helper.cs:line 871
   at Warp.TiltSeries.PerformMultiParticleRefinement(String workingDirectory, ProcessingOptionsMPARefine optionsMPA, Species[] allSpecies, DataSource dataSource, Image gainRef, DefectModel defectMap, Action`1 progressCallback) in /home/pranav/Desktop/pranav/software/warp/WarpLib/TiltSeries.cs:line 6382
   at WarpWorker.WarpWorker.EvaluateCommand(NamedSerializableObject Command) in /home/pranav/Desktop/pranav/software/warp/WarpWorker/WarpWorker.cs:line 735
   ```
   
 Btw, each time it appears to be a different idx on which it fails
 run1: 25/41Unhandled exception. System.Exception: BadRequest: System.Net.Http.HttpConnectionResponseContent
 run2:30/41Unhandled exception. System.Exception: BadRequest: System.Net.Http.HttpConnectionResponseContent
 run3:22/41Unhandled exception. System.Exception: BadRequest: System.Net.Http.HttpConnectionResponseContent

alisterburt, 2024-12-11:
thanks @shahpnmlab - does it run to completion if you don't set `--min_particles`?

shahpnmlab, 2024-12-11:
No setting the (i.e. defaulting to 1) results in the same error message but for 2/2 experiments the job died on the same idx

shahpnmlab, 2024-12-12:
I managed to get past this issue by removing all old TS.xml files and re-importing the alignments and performing ctf. FYI, i had already performed the first round of refinement in M, but for some reason things went out of sync. Btw, this approach worked with the conda release (dev31).

Closing this for now.

closed 2024-12-12 by shahpnmlab